emuflight/EmuFlight #2: Why?

NullVoxPopuli, 2019-08-09:
why not PR to betaflight? Isn't collaboration better than forking?
I'm all for improving the simplicity of the configurator, but I couldn't see any specifics on the home page -- what are they?

Quick-Flash, 2019-08-09:
EmuFlight is headed in a different direction than betaflight. Also some healthy competition is good for industry. Also some of the things that we want to do as a software betaflight has no interest in and would not accept our PR. So that's basically why.

NullVoxPopuli, 2019-08-09:
> different direction
> some of the things that we want to do

what are those things?

Quick-Flash, 2019-08-09:
32k, helio imuf filtering, supporting helio products, simplification of the pidloop. I could go on, but I think that this is sufficient.

closed 2019-08-09 by Quick-Flash

dzenbot, 2019-12-21:
Why not requesting ownership of https://github.com/heliorc/imu-f and https://github.com/ButterFlight/butterflight instead?

Is this a fork of Betaflight or Butterflight's?