emuflight/EmuFlight #5: Where is best to give feedback?

fxbisto, 2019-08-22:
Hi devs,

Where is an appropriate place to give feedback on these builds?

Wanted to say great work on the firmware, it's flying very well on my new testing build with a china special OBF4SD. There's a couple of things I wanted to discuss regarding flight behaviours, but on the whole it's very smooth.

Cheers

fxbisto, 2019-08-23:
Sorry, found the answer in the readme on the pc. It didn't show up on my phone for some reason.

"Support and Developers Channel
There's a dedicated Discord chat channel here:

https://discord.gg/gdP9CwE "

I'll close this issue and join the discord.

closed 2019-08-23 by fxbisto

Quick-Flash, 2019-08-23:
If you know of a bug a great place to put it is in here. If it is simply tuning help or questions, this is not the place. The github is for bug reports.