emuflight/EmuFlight #7: RC5: Filter Q & W adjustments via OSD do not align with limits in CLI

fxbisto, 2019-08-25:
Managed to find this whilst troubleshooting a non-flier;
OSD settings still allow the Q to go too high but limit the W to 64.

`# get filter
gyro_filter_q = 300  <---How has this saved!
Allowed range: 0 - 160
Default value: 30

gyro_filter_w = 64  <---Max OSD value
Allowed range: 3 - 512
Default value: 32`

Quick-Flash, 2019-08-26:
thanks for pointing this out. I'll look into this right now!

Quick-Flash, 2019-09-05:
it appears that this is fixed in our newer testing versions!
should be out later this week!

closed 2019-09-05 by Quick-Flash

fxbisto, 2019-09-05:
Fantastic news! Unfortunately my charger went wrong but I will also test as soon as I can

Quick-Flash, 2019-09-05:
new version will be out later this week. so once that comes out this should be fixed