emuflight/EmuFlight #17: Cli hanging and Revolt problem

Suzi-uzi, 2019-09-20:
Hello there!  On the Revolt V1, fc is not communicating with esc's. I've tried all protocols. Also when selecting multishot, the min/max, idle throttle values are the wrong way around in the configurator. 
The cli hangs not allowing access. Can't wait to fly this!! Happy to help in any way I can!
Peace and Love

AndreySemjonov, 2019-09-20:
Is it only with Revolt ? Can check with another board ?

Suzi-uzi, 2019-09-20:
Hello Andrey! Just so you know, I used to be a test pilot for
Helio/Butterflight. Still using 3.5.2 1.08! Give me 5 mins please. I'll try
an Omnibus F4. Peace.

meoso, 2019-09-20:
my helio connection froze a bunch yesterday (since the new configurator and 0.1.0 firmware) but i just replugged each time. i _do_ recall it doing this in the _past_ with BuF, so it could be a bug regression of some type in the actual firmware. does it force a reboot like pegasus did, or no? (should it or no)

Suzi-uzi, 2019-09-20:
Ok. On Omnibus F4 the cli does not hang. All good. So this must be Revolt
related. It hangs on "building autocomplete cache". I don't have another
quad running at the moment, but if it helps I can build and test tomorrow
to see if I get the same fc to esc communication problem on an Omnibus. I
also noticed that the motor test page does not work but I think this is
part of the same problem. Blheli configurator does not open either so I'm
guessing it has something to do with the passthrough? Peace

Suzi-uzi, 2019-09-20:
Just tested it on Omnibus F4 with some motors and no problem! All tones are
correct and motors work in motors tab. So it's Revolt related. Hope this
helps!  Peace

AndreySemjonov, 2019-09-20:
@Suzi-uzi thanks. Will take a look later on my revolt FC how it is. But all my main quads on Helio too. You can be Emuf test pilot too if you want.

@meoso yes there is a issue with freeze. We took existing BuF and IMUF code and try to reanimate it :)) so a lot of work to do, but try our best

AndreySemjonov, 2019-09-22:
@Suzi-uzi I made test on my RevoltOSD and Revolt v2. I do not have same problem.

Suzi-uzi, 2019-09-22:
Ok. Maybe it's just a v1 problem then? It could be a resource mapping issue, but I can't get into cli to fix it! Would it be possible to contact you directly please? I don't use discord you see! Peace

AndreySemjonov, 2019-09-23:
Stupid question. 
Did you select correct TARGET when flashed ? I mean: REVOLT?
Also you didn't paste DIFF/DUMP after flashing? And made DFU flash with erase ?

Suzi-uzi, 2019-09-23:
Yes the target is correct! Full chip erase too. I wasn't able to paste anything in cli as it kept hanging! So far I've tried on Omnibus F4, FrskyF4 with no problems. I've got a revolt v2 somewhere. I'll try it out on that and report back!  Peace

AndreySemjonov, 2019-09-23:
Some guys tested Revolt v1 in discord group and seems everything working ok. Probably ESC problem on your side or Revolt itself? Seems like hardware issue.

madchiller, 2019-09-23:
@Suzi-uzi Hey! Glad to see you! I just tested Revolt V1 with Original bolt 4 in 1 and had no issues. Confirmed motor spins and BLheli 16 suite connects as expected.

Suzi-uzi, 2019-09-23:
Hello guys! Got it working!  Yes! I had to download the configurator again, after removing the old one, and hey presto!  Nice. Now it's time to put it through its paces............... Good to be flying a buf variant again! Peace

closed 2019-09-23 by Suzi-uzi