emuflight/EmuFlight #18: Motor 4 issue in Talon F7 v2 board

lustertone, 2019-09-20:
The Talon F7 v2 board put out by Heli-Nation was updated to add the use of RPM filtering in BF 4.x+ and as a result it created issues with motor resource mapping for motor 4 which renders motor 4 non-functional in the CLRacingF7 target that it uses. There was initially a CLI fix made available for correcting the issue and a permanent fix was added to BF 4.0.3+ code. The fix does not work with BF3.5.x as the additional timing parameters that are required for ESC telemetry over the ESC signal wire do not exist in 3.5.x and I learned from the designer that there is no fix planned for anything older than BF 4.x.

I understand that EMU Flight is based off of BF 3.5 code and therefore this issue may not be fixable without some deeper 3.5 code investigation to compare the changes between BF 3.5 to BF 4.x that fixed the issue.

**Additional context**
Just for reference:

**EMU Motor Resource Defaults**
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B01
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE

**BF 4.x Motor Resource Defaults**
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 C08
resource MOTOR 5 A01
resource MOTOR 6 B09
resource MOTOR 7 C09
resource MOTOR 8 B01

**Timer and related data added to CLI in BF 4.x**

**timer**
timer B03 AF1
pin B03: TIM2 CH2 (AF1)
timer B06 AF2
pin B06: TIM4 CH1 (AF2)
timer B07 AF2
pin B07: TIM4 CH2 (AF2)
timer B08 AF2
pin B08: TIM4 CH3 (AF2)
timer B09 AF2
pin B09: TIM4 CH4 (AF2)
timer A01 AF2
pin A01: TIM5 CH2 (AF2)
timer C08 AF3
pin C08: TIM8 CH3 (AF3)
timer C09 AF3
pin C09: TIM8 CH4 (AF3)
timer B01 AF2
pin B01: TIM3 CH4 (AF2)
 
**dma**
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 1
ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin B03 0
pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0
pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
pin B08: DMA1 Stream 7 Channel 2
dma pin B09 NONE
dma pin A01 0
pin A01: DMA1 Stream 4 Channel 6
dma pin C08 0
pin C08: DMA2 Stream 2 Channel 0
dma pin C09 0
pin C09: DMA2 Stream 7 Channel 7
dma pin B01 0
pin B01: DMA1 Stream 2 Channel 5

AndreySemjonov, 2019-09-20:
Need to take a look on it.
Is there a ticket with this issue in BF git?

lustertone, 2019-09-20:
Ticket 7722 covers the initial bug report for BF 4.0 and a permanent fix was added in BF 4.0.3+

Quick-Flash, 2020-02-25:
will be 100% fixed in 0.3.0

closed 2020-02-25 by Quick-Flash

karmakillz1972, 2021-04-24:
im brand new to the drone industry and im having the motor four problem as well with my rooster trying to fly the fusion v2 with the gigawatt esc. any help would be greatly appreciated.