emuflight/EmuFlight #22: Dual RX input

F2GD, 2019-09-25:
The ability to have 2 serial RXs connected simutaniously, with 2 rssi channels (can be shown on osd), and a mode to swap between which rx is the one thats actively used to control the quad. 

for example:

Crossfire1 wired up to uart 1
Crossfire2 wired up to uart 2

Rssi channel = aux 8

Rssi1 on osd
Rssi2 on osd

* next to which is active, or have the inactive one flash

Mode RX_2 switches to Crossfire2 when active

gretel, 2019-10-26:
interesting idea, but has lots of implications which prevents us from implementing this at the current point in time. closing for now..

closed 2019-10-26 by gretel

F2GD, 2019-10-26:
Can the dual rssi be added if an FC has a second adc for inav airspeed?