emuflight/EmuFlight #26: RSSI sticks to channel 0 on F4S butterflight build 3.6.7

BeansFPV, 2019-09-29:
In CLI and configurator, selecting any channel for RSSI always switches back to disabled/channel 0. RSSI on OSD is connected to throttle channel. Analog RSSI is turned on in the Config tab. I can see the RSSI signal on Receiver tab so it too is working.

Came from working XM+ setup on Betaflight 3.2x and 4.0 so I know it works ;) First github issue report.

Quick-Flash, 2019-09-30:
have you tried emu 0.1.0 just to comfirm this bug?

BeansFPV, 2019-09-30:
I have not yet but will try and report back. Buddy said he preferred this build's flight characteristics so I didn't bother upgrading to emu build yet.

BeansFPV, 2019-10-13:
I can confirm it stays on emu 0.1.0 with the same behavior. I can see the RSSI varying on AUX 4 (8ch RSSI firmware on the XM+) but it still does not save the RSSI channel in the configurator (emuconfig) or the CLI.

gretel, 2019-10-26:
@BeansFPV thanks for getting back to us! is the firmware file you flashed `XM+FCC170313-RSSI8.frk` ?

BeansFPV, 2019-10-31:
@gretel Yes, I checked on the Taranis SD card to be sure. That's the one.

nerdCopter, 2020-02-28:
does this help? https://youtu.be/x-qdaVpmn5o

closed 2020-03-26 by gretel