emuflight/EmuFlight #30: irratic loop times on Lumenier micro lux v3

Donpilkerton, 2019-10-10:
I flashed over to emu flight yesterday and took it for a flight did great, I wasn't paying attention to the cycle time or anything like that. I was resetting my PID profiles today and just noticed that the cycle times are pretty sporadic I am new to emu flight and IMU filtering so this may be normal. It bounces between 250 and 750 on the cycle times. 

**To Reproduce**
Any flash from release firmware for lux4osd
**Expected behavior**
Does not seem to affect flights any I was just curious if this is normal. 
**Flight controller configuration**
```# diff

# version
# EmuFlight / LUXF4OSD (LUX4) 0.1.0 Sep 16 2019 / 22:43:03 (3956d2f50) MSP API: 1.40

board_name 
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature TELEMETRY
feature ANTI_GRAVITY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map TREA1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 900 1625 0 0
aux 1 1 0 1850 2100 0 0
aux 2 13 0 900 1250 0 0
aux 3 26 2 900 1075 0 0
aux 4 28 2 900 1100 0 0
aux 5 35 0 900 1250 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 90
set ibata_scale = 4000
set small_angle = 180
set pid_process_denom = 1
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2113
set osd_rssi_pos = 2136
set osd_vtx_channel_pos = 2101
set osd_avg_cell_voltage_pos = 2081
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_freq = 5917
set vcd_video_system = NTSC

# profile
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain = 5000
set feedforward_transition = 10
set iterm_relax = RP
set throttle_boost = 50
set i_decay = 7
set p_pitch = 63
set i_pitch = 70
set f_pitch = 100
set p_roll = 55
set i_roll = 70
set d_roll = 30
set f_roll = 100
set i_yaw = 70
set f_yaw = 100

# rateprofile
rateprofile 0

set roll_rc_rate = 126
set pitch_rc_rate = 126
set yaw_rc_rate = 115
set roll_expo = 35
set pitch_expo = 35
set yaw_expo = 35
set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 76```

**Setup / Versions**

# EmuFlight / LUXF4OSD (LUX4) 0.1.0 Sep 16 2019 / 22:43:03 (3956d2f50) MSP API: 1.40
Lumenier micro lux v3 stack with Lumenier 30A BLHeli_32 4in1 ESC (2-4s)
akk fx3 vtx wired directly to vBat on esc, runcam split mini 2 powered from vtx, all using onboard osd, on the flight controller. 
Add any other context about the problem here.

Quick-Flash, 2019-10-12:
I think that this is normal. Type tasks in the cli multiple times just to verify that your pidloops are normal.

Donpilkerton, 2019-10-12:
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       1       1    0.0%    0.0%         0
01 - (         SYSTEM)    984       4       0    0.3%    0.0%         5
02 - (       GYRO/PID)   7812      57      46   44.5%   35.9%      4707
03 - (            ACC)    984      16       8    1.5%    0.7%       125
04 - (       ATTITUDE)    391      18      11    0.7%    0.4%        60
05 - (             RX)     32      59      51    0.1%    0.1%        27
06 - (         SERIAL)     99  313231       1 3100.9%    0.0%       603
08 - (BATTERY_VOLTAGE)     49       5       2    0.0%    0.0%         1
09 - (BATTERY_CURRENT)     49       4       1    0.0%    0.0%         0
10 - ( BATTERY_ALERTS)      4       5       1    0.0%    0.0%         0
11 - (         BEEPER)    100       6       1    0.0%    0.0%         1
18 - (      TELEMETRY)    247     635       0   15.6%    0.0%      1136
21 - (            OSD)     59     466      44    2.7%    0.2%        33
23 - (            CMS)     59       4       1    0.0%    0.0%         0
24 - (        VTXCTRL)      4      21       4    0.0%    0.0%         0
25 - (        CAMCTRL)      4       3       1    0.0%    0.0%         0
27 - (    ADCINTERNAL)      0       2       0    0.0%    0.0%         0
28 - (       PINIOBOX)     19       4       1    0.0%    0.0%         0
RX Check Function                   4       2                         0
Total (excluding SERIAL)                        65.4%   37.3%
PID/GRYO seems to be on point pretty close to 8k. I just wanted to make sure!!!! Thanks!

closed 2019-10-16 by Quick-Flash