emuflight/EmuFlight #34: OSD Hangups during PID tuning

N4V1G4T0R, 2019-10-26:
Hello,
first, great work guys. An ultralight build, which had troubles to find the perfect tune, already flew like a rocket with emuflight default settings. And now even better after some fine tuning.

**Describe the bug**
And here is the issue: during tuning via OSD I experienced hang ups. Just switching to the PID profile settings, the OSD just froze. It required a reboot (reconnecting battery) to work again. No hangups during flight though. Unfortunately, editing settings via LUA Script does not work (BF Lua). This would be a nice alternative to OSD tuning.

**Flight controller configuration**
Attached, you will find the diff.
[diff_n4v1_ultralight.txt](https://github.com/emuflight/Pegasus/files/3775115/diff_n4v1_ultralight.txt)

**Setup / Versions**
The setup is an ultralight build (210g, 6S, CrissCross 5" Frame, AIKON AK32 Flytower MINI STM32F405 Flight Controller & 35A Blheli_32 ESC Combo, Hyperlite HV Motors, TBS Crossfire Nano, TBS Unify Nano). Emuflight 0.1.0 Sep 16 2019.

Regards
N4V1

**To Reproduce**
Startup Configuration via OSD. Go into the PID profile Menu, edit settings, save, fly. Do this about 3 times w/o disconnecting battery and the Configuration OSD may freeze.

**Expected behavior**
no freezes

gretel, 2019-10-26:
@emuflight/emuteam ping

Quick-Flash, 2019-10-29:
could you try turning telemetry to off? right now telemetry is a known to have bugs. Would just like to know if these two events are related.

N4V1G4T0R, 2019-10-30:
Tested it with Telemetry off. No change. Still having the hangups. Even with a different FC (airbot nano v6, target OMNIBUSF4FW instead of Aikon F4, Target AIKONF4).

gretel, 2019-10-30:
@N4V1G4T0R while not related to your isse - nice rig :)

N4V1G4T0R, 2019-11-02:
@gretel thanks :-) And it's flying great with emuflight.

gretel, 2019-11-03:
@N4V1G4T0R  could you please ⚠️remove the propellers and try the **experimental** firmware files attached? please unzip prior to flashing.

[EmuFlight_0.1.21_OMNIBUSF4FW.hex.zip](https://github.com/emuflight/EmuFlight/files/3801913/EmuFlight_0.1.21_OMNIBUSF4FW.hex.zip)
[EmuFlight_0.1.21_AIKONF4.hex.zip](https://github.com/emuflight/EmuFlight/files/3801914/EmuFlight_0.1.21_AIKONF4.hex.zip)

N4V1G4T0R, 2019-11-03:
@gretel the omnibus (airbot nano v6) was testes successfully with the attached experimental firmware files. No hangups anymore during PID tuning via OSD. 

Just some side effects: motor beeper is not working, the configurator sometimes hangs when switching from PID to another tab, the default voltage settings (min 7,4V max 17,4 etc. are a litte off), and no gyro is shown (know issue). I guess, these side effects come from the experimental nature of the firmware. But maybe it's of interest.

If you need me to try it  on the AIKON F4 as well, let me know.

gretel, 2019-11-03:
@N4V1G4T0R @emuflight/emuteam  i'm happy you are getting not locked-up anymore. 

yes, we can only focus on small steps at a time, being a very small team.

can you please open seperate issues here in regards to the voltage settings and on the configurator over at https://github.com/emuflight/EmuConfigurator/issues - thank you very much.

N4V1G4T0R, 2019-11-03:
Don't worry, the default voltage settings a for the official release is fine, it's just the experimental code. Same counts for the motor beeper.

gretel, 2019-11-03:
guess @Quick-Flash will fix this soon - i just took the current state to allow you to check. can you please open issues for the beeper and the hanging configurator?

gretel, 2019-11-04:
upcoming release does include related fixes.. closing for now.

closed 2019-11-04 by gretel

N4V1G4T0R, 2019-11-15:
Just a remark, being unsure if it's possible or helpful to comment on closed issues:

I tried a HolyBro Kakute F7 with that rig. Default Font. There were no OSD freezes anymore, but the numbers (not the letters) disappeared at about the same time, when the OSD on other FCs froze. So, there might be a relation between the freezes and the fonts.

gretel, 2019-11-15:
@N4V1G4T0R also noticed symbols garbling up.. but no freezes for me. while ugly looking these might work: https://github.com/betaflight/betaflight-configurator/tree/master/resources/osd/1

KAJE1977, 2020-03-30:
the butterflight lua scripts work with emuflight

KAJE1977, 2020-03-30:
not everything is in there but enough for the basics. maybe someone can rewrite them for more functionality if they are up for it. i havent got a clue. just stumbled upon it.