emuflight/EmuFlight #36: Telemetry Display

N4V1G4T0R, 2019-10-28:
**Describe the bug**
This is only a minor bug. The VBAT is shown as "RxBt" in the Telemetry Data.

**To Reproduce**
Activate Telemetry in the Config Tab of the emuflight Configurator (0.2.0). On the Transmitter (FrSky X-Lite mit Crossfire Stealth Mod). Search for Telemetry Data ("Discover New Sensors"). VBAT is shown under 10:RxBT (and it takes very long to show the correct Voltag)

**Expected behavior**
Display as VBAT. It is quite scary to show 24V RxBT when the RX is a XF nano (operating Voltage 5V).

**Flight controller configuration**
[EMUF_cli_N4V11_XXt_20191028_200751.txt](https://github.com/emuflight/EmuFlight-Butter-Varient/files/3780261/EMUF_cli_N4V11_XXt_20191028_200751.txt)

**Setup / Versions**
The setup is an ultralight build (210g, 6S, CrissCross 5" Frame, AIKON AK32 Flytower MINI STM32F405 Flight Controller & 35A Blheli_32 ESC Combo, Hyperlite HV Motors, TBS Crossfire Nano, TBS Unify Nano). Emuflight 0.1.0 Sep 16 2019.
Transmitter is a modified FrSky X-Lite with a TBS Crossfire Micro Module

**Additional context**
Also, the amperage is off quite a bit. But that may be just a matter of calibration.

gretel, 2019-10-29:
@N4V1G4T0R thanks for the report! will look into.

> Also, the amperage is off quite a bit. But that may be just a matter of calibration.

in most cases this is true - can you calibrate please?

N4V1G4T0R, 2019-10-30:
I can't test the calibration atm because I switched the FC, which does not have a distinct pad for the current.
However, ESC Sensor does not work either for current measurement. It just shows zero. Same counts for the voltage. ESC Temperature works. And RPM somehow: it only shows one RPM instead of four in the OSD. So, no current measurement for me atm. Who needs that anyway. ;-)

gretel, 2019-10-30:
@N4V1G4T0R on a recent build i had to swap cables on the fc-esc connector cause it didnt match initially..

@Quick-Flash `blheli32` rpm telemetry should be supported?

N4V1G4T0R, 2019-11-02:
@gretel I could try soldering the telemetry signal to a different Uart. At the moment, it's on the default UART (standard connector of the stack from ESC to FC, UART4). However, because ESC temperature and ESC RPM work, I assume the connection is functional, basically. Not sure if the resoldering will change anything, tbh.

Quick-Flash, 2019-11-02:
@gretel i think that blheli32 rpm telemetry works. I have never used it and we may have a bug keep us from getting it to work correctly.

gretel, 2020-04-01:
works for me, `ori32` and `heliospring`.

closed 2020-04-01 by gretel