emuflight/EmuFlight #37: Diatone GT-M530 OSD not working in emu, works in betaflight.

GlossyFPV, 2019-10-29:
OSD disappears after Emuflight furyf4osd 0.1.0. install. OSD works fine with Betaflight. Ive compared cli dumps from bf and emu with no major differences.

**To Reproduce**
install emu on a diatone gt-m530 and the osd disappears, the only way to get it back is to reinstall betaflight and disconnect the vtx and cam from fc. once reconnected osd is back. 

**Expected behavior**
OSD should work like it did in betaflight.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

```# diff

# version
# EmuFlight / FURYF4OSD (FY4O) 0.1.0 Sep 16 2019 / 22:29:05 (3956d2f50) MSP API: 1.40

board_name 
manufacturer_id 

# name
name GT-M530

# resources

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1850 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 2 2 1700 2100 0 0
aux 3 13 5 1900 2100 0 0
aux 4 26 6 1350 1625 0 0
aux 5 31 6 1875 2100 0 0
aux 6 35 4 1825 2100 0 0
aux 7 36 1 1900 2100 0 0
aux 8 39 3 1875 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set rssi_channel = 16
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set ibata_scale = 255
set small_angle = 180
set pid_process_denom = 1
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1000
set osd_vbat_pos = 2433
set osd_rssi_pos = 2050
set osd_tim_2_pos = 2455
set osd_throttle_pos = 2181
set osd_vtx_channel_pos = 2392
set osd_current_pos = 2443
set osd_mah_drawn_pos = 2253
set osd_craft_name_pos = 2069
set osd_pidrate_profile_pos = 410
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2341
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_freq = 5917
set vcd_video_system = NTSC

# profile
profile 0

# rateprofile
rateprofile 0

set thr_mid = 25
set thr_expo = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 78
set tpa_breakpoint = 1650 ```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; Mamba f405 v1 stack
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
XM+, TBS 800mw unlabeled, runcam swift
 - how are the different components wired up.
VTX to fc. CAM to fc. 

**Additional context**
please read emuflight discord for additional info. look for glossy_fpv.

gretel, 2019-10-30:
@GlossyFPV thanks for the report! my main rig has a `mamba f405 mini v2` stack - OSD is working fine for me on both.

did you do a `full flash erase` on flashing emu?

GlossyFPV, 2019-10-30:
Yes I have 4 mamba stacks. No issues with any. I always full chip erase on reflash.  I've reflashed between bf and emu at least 10 times. This is the issue. The gt m530 has a V1 f405 stack. All my other stacks are mk11. I have mamba f405 mini stacks and f405 mk11 working with no problem.  I've narrowed it down to the only rig with a v1 board. It's deffinetely a bug.Thanks,Mark

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/emuflight/EmuFlight-Butter-Varient/issues/37?email_source=notifications\u0026email_token=ANRXQG6GQU67XTIJ3BTXGQLQRIBGZA5CNFSM4JGPSSVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECV6BIY#issuecomment-548135075",
"url": "https://github.com/emuflight/EmuFlight-Butter-Varient/issues/37?email_source=notifications\u0026email_token=ANRXQG6GQU67XTIJ3BTXGQLQRIBGZA5CNFSM4JGPSSVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECV6BIY#issuecomment-548135075",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

GlossyFPV, 2019-10-30:
If you go on the discord: emuflight tab, you'll see I've tried everything I can with the dev's and they even say it's a bug. I've compared cli dumps between bf and emu to see where it went wrong and theres no differences. I've tried everything they've asked and even they are stumped. Thanks,Mark

gretel, 2019-10-30:
@GlossyFPV thanks.. please try to not to quote when using email cause it.. looks messy :)
discord is not syncing with github.. i can either work on here.. or the discord.. i choose github.

gretel, 2019-10-30:
@Quick-Flash looks like issue is related to the `mamba f405 v1` stack only.

GlossyFPV, 2019-10-30:
@gretel gotcha i just realized what the email was doing. sorry about that.  osd works fine with betaflight so im not sure the connections are the issue. plus its worked fine out of the box since ive got it. Bind and Fly. ive gone over all connection with a multimeter as i always do with every quad. everythings as it should

GlossyFPV, 2019-10-30:
i believe @Quick-Flash is the one who sent me here to post. @gretel

gretel, 2019-10-30:
@GlossyFPV we work on the same team as developers and will look into it. thanks again

gretel, 2019-11-03:
@GlossyFPV could you please ⚠️remove the propellers and try this **experimental** firmware? please unzip prior to flashing.

[EmuFlight_0.1.21_FURYF4OSD.hex.zip](https://github.com/emuflight/EmuFlight/files/3801903/EmuFlight_0.1.21_FURYF4OSD.hex.zip)

GlossyFPV, 2019-11-03:
@gretel hey, so i just tried that 0.1.21_furyf4osd firmware with no luck. did a full chip erase like always. i also tried changing the font with no luck.

gretel, 2019-11-03:
@GlossyFPV thanks for trying. works fine for me with the hex attached - i'm sorry to say i have no other idea at the moment.

GlossyFPV, 2019-11-03:
ok thanks anyways, no worries, ill just run betaflight on this one. must just be my board. weird that osd works with betaflight... so does this "bug" get taken down now and not worked on anymore?

gretel, 2019-11-03:
@GlossyFPV i'm just saying me personally can not reproduce this issue right now. same goes for the relation of configuration and firmware in use. we are a very small team..

GlossyFPV, 2019-11-03:
@gretel i feel like an idiot but ive narrowed it down to my cam i think? everytime i unplug the cam from fc the osd comes back. everytime i plug it back in the osd drops out..if you know what this could be please help as im semi new to this hobby and really dont know where to start. im sorry for leading you in the wrong direction with this issue. osd seems to be working on furyf4osd_0.1.0 when my cam is unplugged. thanks for your help.

GlossyFPV, 2019-11-03:
just swapped out cam with one i had lying around. cam was my issue. like i said im sorry for adding more issues to your plate when it wasnt even anything you could do. good luck with future development and i appreciate everything you devs do!! this bug can be closed.

gretel, 2019-11-03:
@GlossyFPV you dont have to feel like that cause you got back to us! it's hard to debug issues like you got.. timing can be different.. few miliseconds are a huge timespan these days...

cool we figured this is out! closing for now ... @emuflight/emuteam

closed 2019-11-03 by gretel