emuflight/EmuFlight #38: No Gyro detected on Airbot Nano V6 default settings

N4V1G4T0R, 2019-10-30:
**Describe the bug**
After Flashing the Airbot Nano V6 with OMNIBUSF4FW/OBFW there is no gyro detected.

**To Reproduce**
Flash the FC with the target. Start the configurator and you will see that the gyro is not shown.

**Expected behavior**
Gyro should be detected with the default settings.

**Workaround**
CLI command "set gyro_to_use = SECOND" will do the trick. Also, you need to use the Roll, Pitch and Yaw Degrees in the Config Tab to change the orientation, instead of the Gyro Alignment.

**Flight controller configuration**
[EMUF_cli_N4V11_XXt_20191030_214501.txt](https://github.com/emuflight/EmuFlight-Butter-Varient/files/3791088/EMUF_cli_N4V11_XXt_20191030_214501.txt)

**Setup / Versions**
The setup is an ultralight build (210g, 6S, CrissCross 5" Frame, Airbot Nano V6, Aikon 4in1 35A Blheli_32 ESC Combo, Hyperlite HV Motors, TBS Crossfire Nano, TBS Unify Nano). Emuflight 0.1.0 Sep 16 2019.

**Additional context**
Afaik, this bug was present in betaflight at some version.

gretel, 2019-10-30:
@N4V1G4T0R thanks for posting the fix! we will have that default...

Quick-Flash, 2020-02-25:
kakutef7HDV has its own target now

closed 2020-02-25 by Quick-Flash