emuflight/EmuFlight #39: No Gyro detected on KAKUTEF7 HDV : EmuFlight / KAKUTEF7 (KTF7) 0.1.0

Metalic91, 2019-10-31:
**Describe the bug**
After Flashing the KAKUTEF7 HDV with KAKUTEF7 firmware there is no gyro detected.

**To Reproduce**
Flash the FC with the target. Start the configurator and you will see that the gyro is not shown.

**Expected behavior**
Gyro should be detected with the default settings and virtual qwad must move

**Flight controller configuration**
```
# 
# diff
# version
# EmuFlight / KAKUTEF7 (KTF7) 0.1.0 Sep 16 2019 / 22:37:10 (3956d2f50) MSP API: 1.40
board_name 
manufacturer_id 
# name
# resources
# mixer
# servo
# servo mix
# feature
# beeper
# beacon
# map
# serial
serial 0 1 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# master
set motor_pwm_protocol = DSHOT1200
set pid_process_denom = 1
# profile
profile 0
# rateprofile
rateprofile 0
```
**Setup / Versions**
 - Flight controller Kakute F7 HDV - Holybro + ESC 4en1 Tekko32 F3 HDV 40A - Holybro (STUDIOSPORT)
 - RX R9 MM welde on the board
 - VTX : DJI air unit HD FPV system
 - 5" Frame Iflight DC5 HD/ Xnova motor hardline (total build 450g)

**Additional context**
On betaflight 4.0.6 i have the same issue with firmware KAKUTEF7(HBRO), no gyro.
It work fine with KAKUTEF7 HDV(legacy) firmware, gyro is ok with this firmware,i'm flying with this actually.

Metalic91, 2019-10-31:
For Information i try to re install the firmware KAKUTEF7(HBRO) on betaflight and propose me default instalation and in CLI i see the version STM32F745 and with that the gyro is ok.
version
# Betaflight / STM32F745 (S745) 4.1.0 Oct 16 2019 / 11:58:25 (c37a7c91a) MSP API: 1.42
# manufacturer_id: HBRO   board_name: KAKUTEF7   custom defaults: YES

gretel, 2019-10-31:
@Metalic91 thanks for the report. can you please try

`set gyro_to_use = SECOND`

on the cli and see if the gyro gets detected? we had a pilot reporting this as a workaroung at https://github.com/emuflight/EmuFlight-Butter-Varient/issues/39

Metalic91, 2019-10-31:
Hello gretel,
Already done but command doesn't work.
# set gyro_to_use = SECOND
###ERROR### Invalid name

gretel, 2019-10-31:
sorry i was pointing to the same issues.. https://github.com/emuflight/EmuFlight-Butter-Varient/issues/38
strange it seems to work in another case..

goldsyndicate, 2019-11-03:
same here , betaflight flashes fine. gyro works , everything is good on beta, but emu will flash and it does not detect gryo - NO_GYRO for arming flag

gretel, 2019-11-04:
we are working on fixing this issue, next release is upcoming. thanks for your patience.

Metalic91, 2019-11-04:
ok thanks, if you want more datas (dump or other, i have 3 kakute F7) don't hesitate i will be happy to help you

goldsyndicate, 2019-11-04:
My kakute also was not showing an accelerometer as well if that helps ,
always down to help any way I can , emu flies a million times better than
any other flight controller. Thank you for all that you are doing let me
know if I can help in any way

-- 
Best,

*Brandon Gold*

*All Things Vape*

*CEO & Founder*
17 Farmington Ave
Plainville, CT 06062
860-540-3680
allthingsvape.com
medvapesupply.com

vltluigi, 2019-11-07:
Same problem on my KakuteF7 1.5 did someone find a way to fix it ?

vltluigi, 2019-11-07:
If you need any info / dump let me know

gretel, 2019-11-07:
> we are working on fixing this issue, next release is upcoming. thanks for your patience.

goldsyndicate, 2019-11-20:
are there any updates on timeframe for a new .hex for the kakute f7. I would really LOVE to have emu flight running on all my rigs . thanks to all the devs for all that you do!

FreerideFPV, 2019-11-28:
Same problem with my kakute F7 HDV. When you release the new firm? I'm interest for helping you to develop some new features and trying some tests for you. I'm flying with the new DJI FPV system and I have some other quads on analog too.

dronexmark, 2019-12-18:
Any date on the kakute f7hdv no gyro fix please?

gretel, 2019-12-19:
2020 maybe.

dronexmark, 2019-12-19:
That's not a month it's a year😯 think I best change my fc, are you having any other boards experiencing problems?

gretel, 2019-12-19:
actually, it's 13 days left in 2019.

dronexmark, 2019-12-19:
I understand that, but I was asking for a month like January or February? not just 2020, because that could be months and months away. Thank you for your reply and I understand it's not an easy fix or it would be done already.

gretel, 2019-12-19:
you are welcome to help doing the development work required!

dronexmark, 2019-12-19:
I would gladly help if Its something a none coder can do?

Quick-Flash, 2019-12-19:
For anyone interested its already in the code. Been added for about a week now.
[image]

dronexmark, 2019-12-19:
I must be blind, I looked last night and didn't   see it, I will try again later. Thank you very much

gretel, 2019-12-19:
so thats negative two weeks ;)

Sent from your iPhone

dronexmark, 2019-12-19:
Thats good to here. I'm on that as soon as I finish work

dronexmark, 2019-12-19:
I'm confused, I thought I missed it last night but I've checked again and HDV isn't in the targets?.I'm on 0.2.12 configurator, any idea's?

FreerideFPV, 2019-12-19:
I've just flash mine and it works great

dronexmark, 2019-12-19:
Me too 😁 just need test fly now

Metalic91, 2019-12-20:
Hello thank lot team gyro work !!
Due to weather it's not possible to fly, i will be back you after firt flying test.
have nice Christmas holiday

dronexmark, 2019-12-20:
Quick question, is a higher Q value mean more filtering or less?

Quick-Flash, 2019-12-23:
@dronexmark higher q is less filtering

closed 2019-12-23 by Quick-Flash

dronexmark, 2019-12-23:
Cheers quickflash

fcfpv, 2020-01-06:
Salut a tous
Je viens demander votre aide 
Impossible d avoir le gyro et l accéléromètre sur ma kakute 7f aio
Pourriez-vous m aidez.merci

FreerideFPV, 2020-01-06:
Si tu as une kakute F7 HDV, il y a une nouvelle target qui fonctionne et
dispo en version release candidate kakuteF7HDV 0.2.0 RC1

fcfpv, 2020-01-06:
Merci freeride.cest ce que j ai fait .je test cette aprem et fais mon retour.

FreerideFPV, 2020-01-06:
Fais attention j'ai dû remapper les moteurs car ils étaient mal attribués

Ares103, 2020-04-17:
Can someone explain why I have no OSD on this FC (Kakute F7 HDV Holybro) on the 0.2.0 RC1 emuflight version ? I tried to reflash it but always the same. Everything else works, motors are well mapped. I'm on DJI HD 0500 version (the latest, with home direction/GPS).

Ares103, 2020-04-17:
I tried to flash the same card on Betaflight 4.1.2, same machine/configuration and the OSD works fine there….

FreerideFPV, 2020-04-17:
It's normal this fc is for dji fpv so she hasn't a betaflight osd chip and
for this fc you must select a non official target release candidate called
kakutef7hdv

Ares103, 2020-04-17:
I selected this one (I read this entire post) :

FreerideFPV, 2020-04-17:
So what is your problem? Betaflight osb didn't work?

Ares103, 2020-04-17:
Yep, no OSD in Emuflight 0.2.0 RC1 (see post of quick-flash/19.12.2019 above) but it works when I flash the same card/quad on BF 4.1.2

Ares103, 2020-04-17:
I also tried this in the CLI of emuflight but without sucess : 
set max7456_spi_bus = 0

(see : https://www.youtube.com/watch?v=_EEoW1pXgVU )

FreerideFPV, 2020-04-17:
No problem for me you have flash it with erase chip ?

Ares103, 2020-04-17:
Yes

Ares103, 2020-04-17:
So your OSD works on the same card with the same DJI version 0.500 ?

Ares103, 2020-04-17:
> 
> […](#)

[image]

FreerideFPV, 2020-04-17:
Mine works great without make the fix for the loop pid freq

Ares103, 2020-04-17:
Ok, thanx. So I'll try to reinstall the whole thing clean and see if it works...

FreerideFPV, 2020-04-17:
Yes try it. Thats works fine for me and my friends who has the same fc on
emuflight

Ares103, 2020-04-17:
I run out of solutions and must go on betaflight (wich annoys me a lot, as I wanted to test emuflight).
I reinstalled/flashed the whole thing several times but the OSD still doesn't work in emuflight 0.2.0.
Maybe one command in here...?

displayport_max7456_col_adjust = 0
Allowed range: -6 - 0

displayport_max7456_row_adjust = 0
Allowed range: -3 - 0

displayport_max7456_inv = OFF
Allowed values: OFF, ON

displayport_max7456_blk = 0
Allowed range: 0 - 3

displayport_max7456_wht = 2
Allowed range: 0 - 3

esc_sensor_halfduplex = OFF
Allowed values: OFF, ON

esc_sensor_current_offset = 0
Allowed range: 0 - 16000

led_inversion = 0
Allowed range: 0 - 7

dashboard_i2c_bus = 1
Allowed range: 0 - 4

dashboard_i2c_addr = 60
Allowed range: 8 - 119

camera_control_mode = HARDWARE_PWM
Allowed values: HARDWARE_PWM, SOFTWARE_PWM, DAC

camera_control_ref_voltage = 330
Allowed range: 200 - 400

camera_control_key_delay = 180
Allowed range: 100 - 500

Ares103, 2020-04-17:
Finally found it and it was simple : 
when you flash in BF, the UART1 port (DJI data trafic) enables automatically.
when you flash with emuflight, you must enable it manually (in the port tab) and I simply forgot it.
Quite a stupid mistake, I apologize ;)
Thanx guys for your disponibility and service. 
Let's try this emuflight 0.2.0 ! ;)

zippomaker, 2021-02-13:
hello everyone, I tried now to install Emuflight on Kakute F7 (13/02/2021) and I don't have gyro data and accell data, I tried with Emu 0.3.3 and 0.3.2 same result, with betafligfht it works fine, some suggestions ? thank you

nerdCopter, 2021-02-13:
There are diffent verisons for Kakute F. 
KAKUTEF7              KAKUTEF7HDV           KAKUTEF7MINIV1        KAKUTEF7MINIV2        KAKUTEF7V15
The V15 is 1.5 (what Betaflight named V2)