emuflight/EmuFlight #41: LED_STRIP on "MAMBA F405 mini mk2" not working

gretel, 2019-11-03:
**Describe the bug**
feature LED_STRIP does not work.

**To Reproduce**
turning the feature on and wiring accordingly does not light up any LED connected.

**Expected behavior**
same setup works fine on the big brother.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
MAMBA F405 mini mk2

Quick-Flash, 2019-11-03:
Sounds like a target issue. I'm not going to look into any target specific issues for now. At the moment integrating all the bf targets and updating our drivers is more important and will likely fix this.

gretel, 2019-11-04:
tried syncing target from bigbrother without any effect. postponing..