emuflight/EmuFlight #42: jittering cycle time when TELEMETRY is turned on.

gretel, 2019-11-03:
**Describe the bug**
cycle times (shown in the configurator) jumping from 120 to 1020 every second or so.

**To Reproduce**
enable TELEMETRY feature, watch the cycle time.

**Expected behavior**
constant cycle time.

**Setup / Versions**
MATEKF411RX

gretel, 2019-11-03:
[image]

@bubi-007 do you think this might be related? i don't have the skillset required to figure this out..

bubi-007, 2019-11-03:
i haven't played with telemetry a lot.
i just don't use it at the moment.
but yes - it is a candidate.
another one is known to create jitter on the time axis: led

gretel, 2019-11-03:
@bubi-007 thanks, are there any best practices..tools you know of which might help debugging this?

bubi-007, 2019-11-03:
measuring code exection time by hand i assume.
and then log it manually: debug[0] = deltaTimeInMs * 1000 for example.
the same for jitter in the calling frequency.

gretel, 2019-11-03:
@bubi-007 feel like `jlink` debugging would be helpful to get some measurements during runtime. 

https://github.com/Serj-Bashlayev/STM32_Profiler

bubi-007, 2019-11-03:
looks good to me.
haven't used this.
so ahead and report back.
but i don't think the resulting binary would be flyable due to the overhead.

gretel, 2019-11-03:
@bubi-007 ya it's getting crowded in the flash space... running out of options.

gretel, 2019-11-03:
https://wiki.st.com/stm32mpu/wiki/Valgrind

bubi-007, 2019-11-04:
stm32mpu !?!?
wrong film.
but very interesting platform :-)
would love it to have one!

gretel, 2019-11-04:
ah ya me stoopid.

bubi-007, 2019-11-04:
no, it is the future.

gretel, 2019-11-04:
the not-so-far-future is fixing this bug :)

adrianmiriuta, 2019-11-13:
@gretel 

this is the culprit.
comment out gyroReadTemperature();
it's not used anyway ...

#ifdef USE_TELEMETRY
void subTaskTelemetryPollSensors(timeUs_t currentTimeUs)
{
    UNUSED(currentTimeUs);

    // Read out gyro temperature if used for telemmetry
    gyroReadTemperature();
}
#endif

gretel, 2019-11-13:
@adrianmiriuta did a pull request.. thanks! how did you figure?

Quick-Flash, 2019-11-14:
@adrianmiriuta I would really like to get into contact with you. Any chance that I can do that?

adrianmiriuta, 2019-11-14:
@gretel 
Has the same problem with my branch 6 months ago.
@Quick-Flash 
See Discord message

closed 2019-11-17 by loutwice