emuflight/EmuFlight #43: battery voltage shown at 0.0V when using HV packs

gretel, 2019-11-03:
**Describe the bug**
OSD shows voltage of 0.0V when connecting fully loaded packs.

**To Reproduce**
connect a 2SHV pack and watch the OSD.

**Expected behavior**
battery voltage should be shown correct.

**Flight controller configuration**
workaround is to adjust "max cell voltage" to 4.5V which leads to another issue: the battery meter won't show the correct relations.

**Setup / Versions**
2x MATEKF411RX, FURY4OSD

Quick-Flash, 2019-11-03:
Give 0.1.21 a try. Updates to the battery and adding the motor output limit are included in here. Probably fixes any issues you have.

gretel, 2019-11-03:
testers were reporting it's even "more broken" now, so i am not going to risk it for a probabilty.

gretel, 2019-11-04:
fix is included in upcoming release.. closing for now.

closed 2019-11-04 by gretel