emuflight/EmuFlight #44: flyaway on the default PIDs when running micros (2.5")

gretel, 2019-11-03:
**Describe the bug**
quad skyrockets.

**To Reproduce**
build a micro, keep the defaults PIDs, arm, dip throttle.

**Expected behavior**
default PIDs should not be dangerous.
"workaround" is to set all PID to half of the defaults values and tune from there.

**Setup / Versions**
FURY4OSD

Quick-Flash, 2019-11-03:
Wouldn't call this a bug. Just means that default tunes dont work for every craft. Try test build 0.1.20 and see if it helps at all.

bubi-007, 2019-11-03:
a bad tune would cause a flip over of the kwad or it would behave very sluggish in the air.
when it goes straight then it is leveling perfectly and the mixer does it's job as well.
a log would help - as i think it is a windup problem.
too many boosting options.

gretel, 2019-11-03:
@bubi-007 i was suspecting a windup, too - did turn `emuboost` and `throttle boost` off - just the bare minimum was set. i forgot to turn the blackbox on, and will have rain for some time - going to get back when i get to log.

bubi-007, 2019-11-03:
emuboost doesn't cause a windup - throttle as well.
in my case it was an itermboost.
clearly visible in the log file.
that bug pop's up frequently - and thus it should be loggable.

gretel, 2019-11-03:
@bubi-007 not sure what `itermboost` - `TPA` for `I` is below 1 in my case.

bubi-007, 2019-11-03:
airmode

gretel, 2019-11-03:
@bubi-007 are you saying i should try again with `airmode` off?

bubi-007, 2019-11-03:
yes

gretel, 2019-11-03:
https://wiki.st.com/stm32mpu/wiki/Valgrind

Quick-Flash, 2019-12-23:
is this fixed in 0.2.0?

gretel, 2019-12-28:
@Quick-Flash ya

Quick-Flash, 2019-12-30:
brilliant than this issue is fixed!

closed 2019-12-30 by Quick-Flash