emuflight/EmuFlight #50: Tmotor F4 target

GhostMaster9, 2019-11-04:
Hi, can you please add a Tmotor F4 target? 

This is the unified target config used in betaflight, this should help.

https://github.com/betaflight/unified-targets/blob/master/configs/default/TMTR-TMOTORF4.config

gretel, 2019-11-04:
@GhostMaster9 as we currently have a different infrastructure than bf we can not just copy their `cfg` files over.

support got added but is untested and not scheduled for release for now.

GhostMaster9, 2019-11-05:
can i test the target please? thanks

gretel, 2019-11-05:
@GhostMaster9 ok, but we will not be able to fix issues you might have using it. please take the props off! [EmuFlight_0.1.21_TMOTORF4.hex.zip](https://github.com/emuflight/EmuFlight/files/3806964/EmuFlight_0.1.21_TMOTORF4.hex.zip)

GhostMaster9, 2019-11-07:
> @GhostMaster9 ok, but we will not be able to fix issues you might have using it. please take the props off! [EmuFlight_0.1.21_TMOTORF4.hex.zip](https://github.com/emuflight/EmuFlight/files/3806964/EmuFlight_0.1.21_TMOTORF4.hex.zip)

thank you so much, it's working perfectly. smooth as expected from emuflight. :D

gretel, 2019-11-07:
@GhostMaster9 @emuflight/emuteam cool, thanks for the feedback. will be included in the next release!

closed 2019-11-07 by gretel