emuflight/EmuFlight #56: RX loss on MATEKF411RX (MTKS)

N4V1G4T0R, 2019-11-14:
**Describe the bug**
Permanently loosing RX connection. Switching between connection and loss about 2 times a second.

**To Reproduce**
Flash Emuflight 0.1 on a Beta65S and bind D16.

**Expected behavior**
No RX loss

**Flight controller configuration**
[EMUF_cli_N4V8_Ta_20191114_203127.txt](https://github.com/emuflight/EmuFlight/files/3848325/EMUF_cli_N4V8_Ta_20191114_203127.txt)

**Setup / Versions**
 - MATEKF411RX
 - FrSky X-Lite (default, with FrSky XJT D16)
 - This is a BetaFPV Beta65s out of the box

**Additional context**
Tried D8 without success.
Tried set frsky_spi_tx_id = 35,28
set frsky_spi_offset = -26
set frsky_spi_bind_hop_data = 5,143,48,188,93,233,138,43,183,88,228,135,38,178,83,223,128,33,173,78,218,123,28,168,73,213,118,23,163,68,208,113,18,158,63,203,108,13,153,58,198,103,8,148,53,193,98,0,0,0
set frsky_x_rx_num = 2
Reflashing with BF showed same symptoms until rebound, then is was fine again

Quick-Flash, 2019-11-15:
try turning off telemetry.

N4V1G4T0R, 2019-11-17:
Actually, turning off telemetry helps.

However, the copter makes a buzzing noise, even when blocking all four props. But that's another issue. ;-)

closed 2019-12-23 by Quick-Flash

Mihailov-wtf, 2020-11-14:
Have something problem today ) I have no ideal soldering wires to receiver.