emuflight/EmuFlight #63: sudden yaw spins and control loss issue

zvjezduska, 2019-12-18:
**Describe the bug**
There might be two bugs, but are related somehow.
1. When flying agressivelly on 6s, quad makes yaw spins on sharp turns on higher throttle value. 
Sometimes it feels like slow glitch, sometimes quad feels like stuck in mud and feels drifting in curves.
2.  If I insist on stressing quad it results in throttle jam on last input, and loss of control on all axes that lasts 0.5-1.5 sec. 

**Flight controller configuration**
default settings emu 0.1.0 and 0.2.0 on imuf 211 and 213

**Setup / Versions**
 - Flight controller [ helio spring mini ];
 - Other components [ aikon 35a mini blheli32 , amaxinno 2207.5 and 2306 1800kv]

**Additional context**
I hope that I found solution, but last testing showed resolving both issues:
1 - on first quad (amaxinno 2306 1800kv) problems of flying away is resolved by flashing prior version of imuf rc9b, on second quad (amaxinno 2207.5 1800kv) setting esc timing to 22 deg and demag to high seems to resolved loss of control issue.

2 - on both quads: yaw jam and drifting is fixed by lowering stock pids by 20%, set emuboost to max 80, and emuboost limit to 5, absolute control and iterm rotation OFF, iterm relax -> setpoint 15. In filter settings Q = 7000, W= 32 , LPF = 90

Quick-Flash, 2020-02-25:
you tried a newer test version?

zvjezduska, 2020-02-25:
Issue resolved.Faulty esc and also failsafes on other quad.I have to update
this

closed 2020-02-25 by Quick-Flash