emuflight/EmuFlight #64: Pid preset not working on FuryF4OSD Target Mambaf405mk2 fc

BrandoFPV, 2019-12-25:
I can select a pid preset and save and when I open the configurator its back at pick a preset.
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

Quick-Flash, 2019-12-30:
it should still ask you to pick a preset, but your tune should be changed to one of the presets.

BrandoFPV, 2019-12-30:
Yes I saw that afterwards I just couldn't figure out how to take the post down. Sorry I'm an idiot

closed 2019-12-30 by Quick-Flash