emuflight/EmuFlight #67: Portable version of Emuflight configurator

szafranski, 2020-01-06:
Installer takes a long time to be installed. Beside that it wou;d be nice if you can use Emuflight without installation - just portable, zip version would be nice - IMO.

bigjoncoop, 2020-01-13:
Is there any methods to access configurator without using a PC? Android, or Raspberry Pi etc...

nerdCopter, 2020-01-23:
@bigjoncoop 
[RaceDroneSoft](https://www.racedronesoft.com/) is currently in development to match dev versions of EmuFlight.
I personally use [BetaFlight CLI](https://play.google.com/store/apps/details?id=org.betaflightcli) when in the field. 

Check if this version works for you: https://dl.bintray.com/emuflight/dev_cfg/0.2.12-174-windows/

I have installed many version in a Win10 pro VM without issue, of course my UAC is and Driver Signing requirement are both disabled.

EazyFPV, 2020-01-27:
I use RaceDroneSoft is costs but works. The UI could be better but it works.

nerdCopter, 2020-02-24:
Found the slow install myself yesterday and researched it:  The issue is inherited from betaflight installer method. https://github.com/betaflight/betaflight-configurator/issues/1834

Essentially, one should uninstall prior versions before upgrading.
For a first time cleanup, uninstall then delete the entire `C:\Program Files\Emuflight\` folder manually.   
Afterward, always use the the `install/uninstall/upgrade` process.

closed 2020-03-26 by gretel