emuflight/EmuFlight #69: bug: 0.2.6 and beyond fails stable mode

nerdCopter, 2020-01-09:
**Describe the bug**
0.2.5 stable-mode works properly
0.2.6 though 0.2.17 stable-mode does not work, possibly pid loop not active.

**To Reproduce**
Prearm.
Arm stable-mode.

**Expected behavior**
throttle should work.

**Actual Behavior**
motors idle speed only

Tested on Helio and FoxeerF405

```
# 

# diff all

# version
# EmuFlight / FOXEERF405 (FXF4) 0.2.12 Jan  2 2020 / 07:57:53 (80059b2ea) MSP API: 1.44

board_name 
manufacturer_id 
mcu_id 0042004f4e48500920303452
signature 

# reset configuration to default settings
defaults nosave

# name
name XL7

# resources

# mixer

# servo

# servo mix

# feature
feature -DYNAMIC_FILTER
feature SOFTSERIAL
feature TELEMETRY
feature OSD
feature ANTI_GRAVITY

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1400 2100 0 0
aux 1 1 1 1875 2100 0 0
aux 2 13 2 1850 2100 0 0
aux 3 26 0 1850 2100 0 0
aux 4 35 3 1875 2100 0 0
aux 5 36 3 1425 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_type = BIQUAD
set gyro_lowpass_hz = 80
set imuf_w = 64
set rssi_channel = 16
set rc_interp = MANUAL
set rc_interp_int = 21
set rc_smoothing_type = FILTER
set rc_smoothing_input_hz = 45
set rc_smoothing_derivative_hz = 100
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_inverted = ON
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1100
set vbat_lpf_period = 30
set ibat_lpf_period = 10
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 2
set osd_rssi_alarm = 45
set osd_tim1 = 2562
set osd_vbat_pos = 2433
set osd_rssi_pos = 2403
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 406
set osd_remaining_time_estimate_pos = 386
set osd_flymode_pos = 2425
set osd_throttle_pos = 397
set osd_vtx_channel_pos = 56
set osd_current_pos = 341
set osd_mah_drawn_pos = 386
set osd_compass_bar_pos = 74
set osd_pid_roll_pos = 301
set osd_pid_pitch_pos = 333
set osd_pid_yaw_pos = 365
set osd_power_pos = 88
set osd_pidrate_profile_pos = 2049
set osd_warnings_pos = 2249
set osd_avg_cell_voltage_pos = 353
set osd_battery_usage_pos = 384
set osd_stat_tim_1 = ON
set osd_stat_endbatt = ON
set debug_mode = GYRO_SCALED
set vtx_band = 1
set vtx_channel = 8
set vtx_freq = 5725
set vcd_video_system = NTSC
set camera_control_internal_resistance = 490

# profile
profile 0

set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 80
set smart_dterm_smoothing = 1
set anti_gravity_gain = 1100
set feedforward_transition = 100
set iterm_rotation = OFF
set iterm_relax = RP
set iterm_relax_cutoff = 10
set i_decay = 6
set p_pitch = 67
set i_pitch = 63
set d_pitch = 44
set f_pitch = 35
set p_roll = 55
set i_roll = 63
set d_roll = 32
set f_roll = 35
set p_yaw = 90
set i_yaw = 63
set d_yaw = 2
set f_yaw = 35
set abs_control_gain = 15

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 5
set thr_expo = 30
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 77
set tpa_rate_i = 110
set tpa_breakpoint = 1100

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

Quick-Flash, 2020-01-10:
both those fc? I've tried those versions on a few fc with no issues... maybe try to do a setup from scratch?

nerdCopter, 2020-01-11:
additional troubleshooting & findings:
- all flashes with full chip erase and paste xact same diff

while on 0.2.6-230 -- additional findings:
- prearm not a factor
- NFE is indeed a factor just as 3dracingman said.
- with NFE on, angle-mode arms properly but motors sound horrible at idle (like gravel), and smooth at any throttle.
- disabled NFE: angle-mode failed.

flashed 0.2.18-250
- pasted exact same diff: angle failed.
- removed ff-transition & abs-control, and enabled iterm-rotation: angle worked!
- returned ff-trans=1.0: angle worked!
- returned disabled iterm-rotation: angle worked!
- returned abs-control=15: angle still worked! WTF?

reflashed 0.2.15-247 with full-chip erase usual.
- pasted my exact same diff: angle-mode failed.

reflashed 0.2.6-230
- angle-mode failed.
- NFE worked, idle sound like gravel
- NFE disabled, angle-mode failed.
- set ff-transition=0, angle-mode failed; returned ff-transition=1, angle failed.
- set abs-control=0, angle failed; returned abs-control=15, angle failed.
- enabled iterm-rotation, angle failed; disabled iterm-rotation, angle failed.
- try all at once: set ff-trans=0 & abs-control=0, and enable iterm-rotation: angle-failed.

reflash 0.2.18-250
- pasted same exact diff: angle-mode fully works.

reflash 0.2.15-247
- pasted same exact diff: angle-mode fails
- cannot reproduce fix

flash 0.2.17-249
- pasted same exact diff: angle-mode fails
- set ff-trans=0 & abs-control=0, and enable iterm-rotation: angle-failed.
- reverted ff-trans, abs-control, iterm rotation: angle failed.

reflash 0.2.18-250
- pasted same exact diff: angle-mode fully works.

conclusions:
- some variable/value/if-then or other code, was introduced in 0.2.6-230 that causes angle-mode failure.
- every version through 0.2.17-249 fails basic angle-mode.
- **0.2.18-250 completely works**, even with reflash and identical diff paste.

nerdCopter, 2020-01-11:
0.2.18 on Helio works as well. closing issue.

closed 2020-01-11 by nerdCopter