emuflight/EmuFlight #71: TMOTORF4 TARGET - NO GYRO DETECTED

goldsyndicate, 2020-01-15:
No Gyro Detected

Configurator: 0.2.12
Firmware: EMUF 0.2.0
Target: TMOTORF

# get gyro
align_gyro = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

gyro_hardware_lpf = NORMAL
Allowed values: NORMAL, EXPERIMENTAL, 1KHZ_SAMPLING

gyro_32khz_hardware_lpf = NORMAL
Allowed values: NORMAL, EXPERIMENTAL

gyro_sync_denom = 1
Allowed range: 1 - 32

gyro_dyn_lpf = 0
Allowed range: 0 - 16000

gyro_lowpass_type = PT1
Allowed values: PT1, BIQUAD

gyro_lowpass_hz = 90
Allowed range: 0 - 16000

gyro_lowpass2_type = PT1
Allowed values: PT1, BIQUAD

gyro_lowpass2_hz = 0
Allowed range: 0 - 16000

gyro_notch1_hz = 0
Allowed range: 0 - 16000

gyro_notch1_cutoff = 0
Allowed range: 0 - 16000

gyro_notch2_hz = 0
Allowed range: 0 - 16000

gyro_notch2_cutoff = 0
Allowed range: 0 - 16000

gyro_calib_duration = 125
Allowed range: 50 - 3000

gyro_calib_noise_limit = 48
Allowed range: 0 - 200

gyro_offset_yaw = 0
Allowed range: -1000 - 1000

gyro_overflow_detect = ALL
Allowed values: OFF, YAW, ALL

gyro_use_32khz = OFF
Allowed values: OFF, ON

gyro_cal_on_first_arm = OFF
Allowed values: OFF, ON

Quick-Flash, 2020-01-16:
what gyro does your target have?

goldsyndicate, 2020-01-16:
MPU6000

Quick-Flash, 2020-01-16:
does this work in betaflight?

goldsyndicate, 2020-01-16:
Yes beta flight has no issue same target TMOTORF4

madchiller, 2020-03-14:
Did you full chip erase when flashing? @goldsyndicate

closed 2020-04-18 by gretel