emuflight/EmuFlight #72: No Gyro detected on Holybro Kakute F7 Flight Controller AIO.

boof64, 2020-01-16:
Аfter flashing the board Holybro Kakute F7 Flight Controller AIO I have the same error.
Issue #39 didn't help me (different target)

REM: after flashing BetaFlight i have the dialog box with question like "Theare are special settings for this board....".

Quick-Flash, 2020-01-16:
try the kakutaf7hdv target and see if that works

boof64, 2020-01-29:
> try the kakutaf7hdv target and see if that works

Thank you! I'll test it

closed 2020-02-03 by boof64

Rennings1991, 2020-05-17:
Did this work? I'm having the same issue with my kakute f7 aio v1.5 and would really like to run emuflight on it. I'm not sure what target I need to flash.

nerdCopter, 2020-05-18:
@Rennings1991 there will be a v1 and v2 kukute in 0.3.0 release

boof64, 2020-05-18:
> Did this work? I'm having the same issue with my kakute f7 aio v1.5 and would really like to run emuflight on it. I'm not sure what target I need to flash.

Yes, its work for me.