emuflight/EmuFlight #77: no gyro detect on 0.2.0 RC1 - CALIB error in OSD

robomaniac, 2020-01-28:
I saw other user have issue with other boards.
issue #38 and issue #39 

I tried 
set gyro_to_use = SECOND
but ###ERROR### Invalid name

please note that if I enable accelerometer I can see my drone move in configurator!

**Describe the bug**
I have emu 0.1.0 on my mobula6 and I can fly no problem.
I flash 0.2.0 and have CALIB error in OSD.
I can not arm. I set moron settings to 200. (gyro_calib_noise_limit = 200)

I did try 0.2.9

**To Reproduce**
here is pastebin of my settings. (pastebin is down) so
https://justpaste.it/648iq

**Setup / Versions**
mobula6 (crazybee f4 lite) with frsky receiver running D8 since can't fly with D16.
0603 motors

robomaniac, 2020-01-28:
I when back to 0.1.0 and tried set gyro_to_use = SECOND and same error.

robomaniac, 2020-01-28:
**set gyro_to_use**  command is for omnibus only

gretel, 2020-04-01:
friend of mine flies recent emu fine on a `mobula6`. suspect a hardware issue. closing for now.

closed 2020-04-01 by gretel