emuflight/EmuFlight #79: Request for aikon F7 support

4dase, 2020-02-02:
Would love to see a aikon F7 fc support for this!

I use kiss and have tried Emu on a old f4 board. I have some F7 I was given to test but never managed to get betaflight to fly like I wanted to and did try emu on that F4 board and was impressed!

great for cinematic flying!

Dave @ hypercine.com

Kaiowarez, 2020-02-05:
4dase I need a dump all from betaflight and I'll see what I can do

4dase, 2020-02-05:
Hey bro! Let me see if i can do that in a bit!

Kaiowarez, 2020-02-05:
Okay cools thanks

4dase, 2020-02-05:
# dump -all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.3 Jan 16 2020 / 11:35:53 (543a5e1ba) MSP API: 1.42
# manufacturer_id: AIKO   board_name: CLRACINGF7   custom defaults: YES

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id AIKO

# name: GU

# resources
resource BEEPER 1 C15
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 B06
resource MOTOR 6 B08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
[... 1000 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 122
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 20
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 60
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

Kaiowarez, 2020-02-05:
try to jump on discord, since this would be experimental in a sense

closed 2020-03-10 by gretel