emuflight/EmuFlight #88: Original Splashscreen OSD ?

ochbob, 2020-02-28:
Hello,

I don't know why, but I have an issue with Emuflight stock OSD Splashscreen.

Where can I found the original one ?

Thank you.

deanrather, 2020-03-05:
I'm not one of the devs but I have been mucking around with fonts and the osd screen a bit.

I think maybe what yo u're looking for is the "default.mdm" from here https://github.com/emuflight/EmuConfigurator/tree/master/resources/osd/2

(or maybe from the adjacent "1" folder)

If you load that in using the "Load Font File" feature on the OSD tab of the configurator, that might fix it.

ochbob, 2020-03-06:
Will try, thank you :)

gretel, 2020-03-10:
just upload any font using the emu configurator to get an emu on ye osd!

closed 2020-03-10 by gretel