emuflight/EmuFlight #95: OSD Cursor Font Invalid

deanrather, 2020-03-05:
**Describe the bug**
I'm new to this, so apologies if I'm off the mark here.
When I turn on the "cursor" (crosshairs) in the OSD, it is comprised of 3 characters. It's grabbing these 3 chars from the wrong positions in the font file.

I think the cause is because emuflight is shipping with the newer font files, but the code is handling them as if it were the older font files.

Here's the betaflight MR where they changed the way to handle the font files.

https://github.com/betaflight/betaflight/pull/3461/commits/8becc3db9f6bfff3b7f2339dda22eaf871c9286f

I suspect I could solve this issue by loading in an older font file, but it would be better to update emuflight to support the newer format.

again, this is 50% guesswork, I might be totally off the mark here :)

**To Reproduce**
- Configure the OSD to have the crosshair
- See that it's not a nice looking crosshair, it's 3 random characters

**Expected behavior**
- The crosshair to be a nice cursor

**Flight controller configuration**
Note that I've loaded a custom font in attempt to fix the issue, so that might be reflected in this diff. either way, it didn't work, the issue existed with the OG font and my custom one :)

```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# EmuFlight / MATEKF411RX (M41R) 0.2.0 Dec 13 2019 / 04:22:28 (c1bd326) MSP API: 1.43

board_name 
manufacturer_id 

# name
name TH2PROSK8R

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -AIRMODE

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 1 900 1300 0 0
aux 1 1 0 1700 2100 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 1 1300 1700 0 0
aux 4 28 0 900 1625 0 0
aux 5 35 0 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_dyn_lpf = 100
set gyro_lowpass_hz = 100
set gyro_lowpass2_hz = 300
set imuf_roll_q = 8000
set imuf_pitch_q = 8000
set imuf_yaw_q = 8000
set imuf_w = 124
set acc_calibration = -365,-60,-57
set min_check = 1076
set max_check = 1976
set fpv_mix_degrees = 21
set rx_spi_protocol = FRSKY_D
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 44
set vbat_full_cell_voltage = 44
set vbat_min_cell_voltage = 31
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2103
set osd_rssi_pos = 2113
set osd_tim_1_pos = 310
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2401
set osd_throttle_pos = 139
set osd_vtx_channel_pos = 2117
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2190
set osd_craft_name_pos = 2433
set osd_pidrate_profile_pos = 2369
set osd_warnings_pos = 2409
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5769
set vcd_video_system = NTSC
set frsky_spi_tx_id = 97,208
set frsky_spi_offset = -101
set frsky_spi_bind_hop_data = 157,212,11,66,121,176,231,30,85,140,195,250,49,104,159,214,13,68,123,178,233,32,87,142,197,252,51,106,161,216,15,70,125,180,235,34,89,144,199,254,53,108,163,218,17,72,127,255,255,255

# profile
profile 0

set dterm_lowpass2_hz = 150
set vbat_pid_gain = ON
set feedforward_transition = 32
set iterm_rotation = OFF
set iterm_relax = RPY
set iterm_relax_cutoff = 12
set throttle_boost = 7
set feathered_pids = 45
set i_decay = 8
set emu_boost = 180
set emu_boost_limit = 35
set p_pitch = 93
set i_pitch = 111
set d_pitch = 87
set f_pitch = 75
set p_roll = 93
set i_roll = 111
set d_roll = 87
set f_roll = 75
set p_yaw = 87
set i_yaw = 100
set d_yaw = 45
set f_yaw = 88
set p_level = 87
set i_level = 87
set d_level = 113
set auto_profile_cell_count = 1

# rateprofile
rateprofile 0

set thr_mid = 33
set thr_expo = 47
set roll_rc_rate = 86
set pitch_rc_rate = 87
set roll_srate = 87
set pitch_srate = 85
set yaw_srate = 111
set tpa_rate_p = 104
set tpa_rate_i = 120
set tpa_rate_d = 115
set tpa_breakpoint = 1450

```

**Setup / Versions**
 - Flight controller It's a Tinyhawk II BNF edition. No mods. Just installed emuflight.

**Additional context**
Love your work :) Hope to contribute someday!

deanrather, 2020-03-05:
here's a lame-ish fix here. it's just a tweak of the existing font file to fix the chars that are rendered for the crosshairs

https://github.com/emuflight/EmuConfigurator/pull/64

Quick-Flash, 2020-03-06:
no thanks for pointing this out. I'm fairly certain that we did apply to a fix to get the bf v2 fonts to work in more recent dev versions. NerdCopter should know more about it as i'm pretty sure he did the code to fix it.

deanrather, 2020-03-06:
good to hear!

closed 2020-03-06 by deanrather

nerdCopter, 2020-03-06:
yessir, v2-font usage are merged in newer dev firmware versions and v2-fonts were merged into newer dev Configurator versions.

If anything, we (i) mistakenly used betaflight v1 fonts to build emuflight logo, whereas emuflight 0.1.0, 0.2.0 was using butterflight font layout that was pre-v1 betaflight font layout.

it was discussed if we should fix v1 fonts to match butterflight layout but was decided to just move forward with v2 fonts and fix firmware side.  this also gave additional characters for things such as GPS usage.

NOW, if someone wants to take a butterflight font, and use it as a "template" to add each of the the script types (default,vision,large,extra_large,impact,impact_mini,digital,clarity,bold,emuflight[duplicate of default]) and Pull Request to EmuConfigurator/resources/osd/1/, than would be useful. 

However, this would be a very time consuming task to manually build each font for the butterflight layout. (one option is to limit this to one or two fonts and call it a day. i.e. clarity & default ?)

here is the butterflight fonts: https://github.com/ButterFlight/butterflight-configurator/tree/master/resources/osd

nerdCopter, 2020-03-06:
this is useful, but has to be modded slightly to use fonts desired. https://github.com/nerdCopter/bflogoreplacer-app

this is useful for creating png from mcm, or reverse: https://github.com/nerdCopter/mcm-tools