emuflight/EmuFlight #96: Request for target: HGLRCF405

MarkiZZZ34, 2020-03-06:
pls add HGLRCF405

Kaiowarez, 2020-03-09:
attach a dump pls

MarkiZZZ34, 2020-03-09:
# 

# dump

# version
# Betaflight / STM32F405 (S405) 4.1.2 Dec 22 2019 / 02:19:35 (89051e256) MSP API: 1.42
# manufacturer_id: HGLR   board_name: HGLRCF405   custom defaults: YES

# start the command batch
batch start

board_name HGLRCF405
manufacturer_id HGLR

# name: MarkiZZZ

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 B05
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B08
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
[... 1023 lines cut ...]
set thr_mid = 27
set thr_expo = 60
set rates_type = BETAFLIGHT
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 65
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 50
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

Kaiowarez, 2020-03-21:
Sorry havent gotten back to you in a bit, can you link me which FC this is?
i started making the target and needed to check the features included with it

MarkiZZZ34, 2020-03-22:
https://www.hglrc.com/products/hglrc-fd435-stack-20x20-3-6s-fd-f4-mini-fc-35a-bl32-4in1-esc

Kaiowarez, 2020-03-22:
Okay thanks will work on it today
If you want to test it when it's done let me know if you have a discord
account I'll send the target hex there

MarkiZZZ34, 2020-03-22:
MarkiZZZ#5107 - discord

Kaiowarez, 2020-03-22:
Cools as soon as I Finish iit I'll ask you to test it

Kaiowarez, 2020-03-23:
Okay done and markizzz says the target is good, please close

closed 2020-03-23 by nerdCopter