emuflight/EmuFlight #118: reflect all settings in OSD

gretel, 2020-03-13:
some settings are not yet available in the OSD. we should have them all, when feature set stabilizes.

Quick-Flash, 2020-03-13:
All settings??? That includes things like this...
[image]

I suggest instead of all settings that we just make a list of the settings that we want and we add that to the osd rather than put all settings... Would take so long to add every single setting. That means that all the receiver stuff would need to be added as well.

Quick-Flash, 2020-03-13:
If we did add all settings to the osd I think we would begin to run into memory issues as well and would really need to just do what falcox does to get things to work...

gretel, 2020-03-13:
sorry i was referring to the PID related stuff only

Quick-Flash, 2020-03-14:
ah makes more sense