emuflight/EmuFlight #143: Need Mode 2 stick functions work on EmuFlight

FrankFPV, 2020-03-26:
Mode 2 stick functions are good for whoops, I hope it can be use in the future with EmuFlight.
Thanks.

gretel, 2020-03-26:
i am flying mode 2, no change required.

FrankFPV, 2020-03-27:
> i am flying mode 2, no change required.

I am saying of mode 2 stick functions
[image]

Quick-Flash, 2020-03-28:
> > i am flying mode 2, no change required.
> 
> I am saying of mode 2 stick functions
> [image]

I am pretty sure that we already do this... I think i have used this before... Can you verify?

gretel, 2020-03-29:
as i said, this is working fine on mode 2.

closed 2020-03-31 by Quick-Flash