emuflight/EmuFlight #148: AKK VTX Bug

chuck-buckets, 2020-04-02:
Just flashed Emuflight today, did a short flight and it is SO much nicer just stock settings than what I could get from hours of tuning in BF. Thank you EmuFlight!!! I am having issues with SmartAudio though.

Essentially what happens is I can get into the menu and select smartaudio, but there are no options in that submenu.

From my research it sounds like it is only a problem with AKK and RDQ vtx's. I have an AKK Ultimate.

It's essentially the exact same problem this guy was having 
https://www.youtube.com/watch?v=fLfXGL23S18 
I guess BF had to release a patch, however, I have not had any problems with SmartAudio in BF 4.0 and up.

```
# version
# EmuFlight / FOXEERF722DUAL (FXF7) 0.1.0 Sep 16 2019 / 22:25:12 (3956d2f50) MSP API: 1.40

board_name 
manufacturer_id 

# name
name Quadiemoto

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 1 2048 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1625 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1725 2100 0 0
aux 3 13 2 1300 2100 0 0
aux 4 35 3 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 18,77,265
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set dshot_idle_value = 1200
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set gps_distance_limit = 1000
set pid_process_denom = 1
set osd_vbat_pos = 2405
set osd_rssi_pos = 2102
set osd_tim_2_pos = 2418
set osd_craft_name_pos = 2090
set osd_warnings_pos = 2442
set osd_avg_cell_voltage_pos = 2411
set vtx_band = 1
set vtx_freq = 5865

# profile
profile 0

# rateprofile
rateprofile 0
```

chuck-buckets, 2020-04-03:
Update. I was shown the legacy_SA_support switch in config. I tried that, but still not able to change anything in smartaudio.

gretel, 2020-04-07:
@chuck-buckets running a `akk fx2` without issues. you sure it's smartaudio, not tramp?

chuck-buckets, 2020-04-14:
@gretel I used smataudio on BF before with this same hardware, and I tried adjusting with tramp but no dice...

gretel, 2020-04-19:
ya it's all broken and won't fix in emuflight for now. lacking resources to backport all the fixes to a different codebase. contributors welcome.

closed 2020-04-19 by gretel

chuck-buckets, 2020-05-16:
I updated to emu 0.2.0 and now it works