emuflight/EmuFlight #160: Is there an Emuflight Target for SucceX Micro F4 FC?

Whatsuppeter, 2020-04-09:
Hello,
I have SucceX Micro F4 flight contoller and the BetaFlight target says "IFRC/IFF4_PRO (STM32F411)", when I plug it into BetaFlight.
Is there an EMUFLIGHT target/Hex file for STM32F411 for my flight controller?  I see there is an STM32F411DISCOVERY hex file.  Can I use this one for my flight controller or is that not the same as STM32F411?
Please let me know.
Thanks!

nerdCopter, 2020-04-09:
We have the following in dev versions:
`IFF411_PRO`
`IFF4_AIO`
`IFF4_TWIN_G`
`IFF7_TWIN_G`
I would assume it's the `IFF411_PRO` in your case. However, this is not yet "released".

Whatsuppeter, 2020-04-09:
OH Great!  I have verified from Iflight that **IFLIGHT_F411_PRO** is the correct target for my FC.  I assume that **IFF411_PRO** is the same one or the one to use for EMUflight.  
Can you tell me when it will be released or if I can obtain and flash an un released version?
Thanks.

nerdCopter, 2020-04-09:
"soon" but soon tends to keep expanding out.  really close in essence.
do note this is github, source code is here.
also consider joining discord server https://discordapp.com/channels/547211754845765635

closed 2020-04-09 by nerdCopter