emuflight/EmuFlight #170: Target Request for LuxMiniF7 board.  tnx.

avivyam, 2020-04-15:
Thank you for contributing to EmuFlight!

## Hardware

If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Any questions on how to report issue?

Please join our discord server:
https://discord.gg/gdP9CwE

## If you are sure there is an issue with the firmware itself please follow these steps:

1. Describe your problem
2. Include ways to reproduce the problem
3. Provide as much information as possible about your hardware and software, including:
 - what hardware / software you are using
 - the version of all the software used
 - how things are connected / wired up
4. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
5. Remove this Text :)

nerdCopter, 2020-04-15:
`LUXMINIF7` exists in the soon to be released 0.3.0; however, it will not be back-ported to 0.2.0.
thank you

closed 2020-04-15 by nerdCopter

avivyam, 2020-04-15:
Thanks man ,

How soon is soon ?

Are you looking for testers ? USAF flight test center graduate ...

A lot of spare time now , forced leave from airline due Covid 19.

I have a set of drones form hx-115 to mb cinematic 7.

  You guys are great by the way

    Stay healthy

     Aviv