emuflight/EmuFlight #178: No OSD in emuflight 0.2.0 RC1 / DJI HD 0500 version (latest) / Kakute F7 HDV Holybro

Ares103, 2020-04-17:
Can someone explain why I have no OSD on this FC (Kakute F7 HDV Holybro) on the 0.2.0 RC1 emuflight version ? I tried to reflash it but always the same. Everything else works, motors are well mapped. I'm on DJI HD 0500 version (the latest, with home direction/GPS).
I tried to flash the same card on Betaflight 4.1.2, same machine/configuration and the OSD works fine there….

gretel, 2020-04-18:
guess `PAL` or `NTSC` is what you mix up.
this is most certainly not related to emuf.

fichek, 2020-04-19:
I don't think DJI cares about PAL/NTSC/AUTO - I get OSD in my DJI goggles on all 3 settings with BetaFlight. Just installed EmuFlight on that quad, configured MSP on same uart, turned on OSD and setup my elements, but they are not showing. Please reopen this issue @gretel

gretel, 2020-04-19:
yes i can reopen but this wont fix for now cause we like one guy coding.

Ares103, 2020-04-19:
UART1 must be enabled in port tab (communication/video datas FC-DJI, not MSP but the next switch). I guess BF does it "randomly¨. That solved my case and OSD/GPS etc work exactly like in BF.

FlightMaster, 2020-04-23:
> Can someone explain why I have no OSD on this FC (Kakute F7 HDV Holybro) on the 0.2.0 RC1 emuflight version ? I tried to reflash it but always the same. Everything else works, motors are well [...]

Hello, just wondering how did you flashed your Holybro Kakute F7 HD to the EMUFLIGHT Configurator?  if so, what is the target firmware for the said Flight Controller above? Thanks in advance

Ares103, 2020-04-24:
Sorry for the delay, had a lot to do. I took the last emuflight configurator 2.1.2 rc2 and there's a target called kakute f7 HDV (see above), select UART1 (tx-rx communication with the DJI module) in ports tab and it works ;)

closed 2020-04-24 by nerdCopter