emuflight/EmuFlight #179: Betafpv AIO 20A v2

LeinesFPV, 2020-04-18:
Hello, everyone,
i have the problem that my Toothpick does not work with the Betafpv AIO 20A v2.
The FC works perfectly with the MatekF411 target and Betaflight.
Unfortunately it doesn't work with EMU.

As an example: I can only arm the quad after the first time I plug in the battery. If I dissarm it then and want to arm it again, it does not work anymore.
The same applies to the OSD. I can only access it once... and then I can't access it again.

https://youtu.be/8C6Qz1FwD9o

If I want to start after the first arm, I have a fligh away...

best regards

Kaiowarez, 2020-04-18:
This problem have been fixed, next release would have the fix

LeinesFPV, 2020-04-18:
Many thanks for the extremely fast answer!

Do you possibly already have a working version? I would love to fly EMU on my Tooth. I have it on all 5" and I love it!

loutwice, 2020-04-19:
i fly emu on my toothpick hx115 without problem

gretel, 2020-04-19:
but he was asking for code, not if it is possible.

closed 2020-04-19 by gretel