emuflight/EmuFlight #210: PID I term problems and drift

yo90bosses, 2020-05-08:
I changed the firmware of my mobula 6 from betaflight to emuflight. On the first flight i had lots on drift towards the CG if it was off center. I didnt have this problem in Betaflight. It also persists If i increase the I term in the PID settings. Im using the settings from this link: https://www.facebook.com/groups/1702300099835884/permalink/2693202914078926/

```
# diff

# version
# EmuFlight / CRAZYBEEF4FS (C4FS) 0.1.0 Sep 16 2019 / 22:04:14 (3956d2f50) MSP API: 1.40

board_name 
manufacturer_id 

# name
name EMU|MOCKULA 6

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature LED_STRIP
feature ANTI_GRAVITY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 1 2048 115200 57600 0 115200

# led
led 0 7,9::A:0
led 1 8,9::A:0
led 2 7,4::CB:1
led 3 8,4::CT:0

# color

# mode_color
mode_color 6 0 5
mode_color 6 1 2
mode_color 7 0 7

# aux
aux 0 0 3 1700 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 1300 1700 0 0
aux 3 35 0 1300 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# rxrange
rxrange 0 988 2010
rxrange 1 988 2012
rxrange 2 988 2012
rxrange 3 988 2012

# vtx

# rxfail

# master
set gyro_lowpass_hz = 200
set gyro_lowpass2_hz = 400
set acc_calibration = 12,-17,66
set fpv_mix_degrees = 20
set airmode_start_throttle_percent = 55
set motor_pwm_protocol = DSHOT600
set bat_capacity = 300
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 31
set vbat_scale = 113
set ibata_scale = 838
set yaw_motors_reversed = ON
set small_angle = 180
set gps_distance_limit = 1000
set pid_process_denom = 1
set osd_rssi_pos = 2231
set osd_tim_2_pos = 2401
set osd_flymode_pos = 2424
set osd_throttle_pos = 321
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2242
set osd_craft_name_pos = 360
set osd_warnings_pos = 2281
set osd_avg_cell_voltage_pos = 2209
set osd_disarmed_pos = 2218
set vtx_band = 1
set vtx_channel = 7
set vtx_freq = 5745
set vcd_video_system = NTSC

# profile
profile 1

set dterm_lowpass_hz = 100
set vbat_pid_gain = ON
set anti_gravity_gain = 1100
set iterm_rotation = OFF
set iterm_relax = RPY
set iterm_limit = 9
set throttle_boost = 4
set i_decay = 7
set emu_boost = 800
set boost_limit = 40
set p_pitch = 75
set i_pitch = 111
set d_pitch = 60
set f_pitch = 80
set p_roll = 75
set i_roll = 111
set d_roll = 60
set f_roll = 80
set p_yaw = 90
set i_yaw = 120
set d_yaw = 0
set f_yaw = 80
set horizon_tilt_effect = 80
set horizon_tilt_expert_mode = ON

# rateprofile
rateprofile 0

set yaw_rc_rate = 204
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 20 
```

iggut, 2020-05-08:
Did you calibrate the acceletometer?

AndreySemjonov, 2020-05-08:
Can you put latest official release and try it ?

nerdCopter, 2020-05-08:
two problems here @yo90bosses :
1) link is to unreleased beta firmware 0.2.15
2) you installed 0.1.0, which will not be supported as it was the first release.

please wait for 0.3.0 which should be soon.

gretel, 2020-05-08:
@nerdCopter you should put up some rules on how you expect people to behave.

nerdCopter, 2020-07-16:
* @yo90bosses drift will be fixed in release 0.3.2 with iDecayV2: https://github.com/emuflight/EmuFlight/pull/320
* if you are stuck on 0.1.0 `set i_decay = 1`

closed 2020-07-16 by nerdCopter