emuflight/EmuFlight #228: Battery average cell voltage is not displayed correctly in the DJI OSD (IFF4_AIO)

mbilozub, 2020-05-18:
**Describe the bug**
The average battery voltage shows 0.0 in the DJI OSD. The total battery voltage looks good.

**To Reproduce**
Enable _Battery average cell voltage_ in the configurator

**Expected behavior**
_Battery average cell voltage_ should show the voltage (like 4.2)

**Flight controller configuration**
```
# version
# EmuFlight / IFF4_AIO (S411) 0.3.0 May 17 2020 / 20:14:47 (c032327bc1) MSP API: 1.47

board_name 
manufacturer_id 

# name
name DJI

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1400 0 0
aux 1 2 1 1700 2100 0 0
aux 2 13 2 1300 1650 0 0
aux 3 35 2 1700 2100 0 0
aux 4 36 3 1700 2100 0 0

# adjrange
adjrange 0 0 4 900 1250 6 5 0 0
adjrange 1 0 4 1300 1700 7 5 0 0
adjrange 2 0 4 1750 2100 8 5 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set acc_calibration = 139,-77,-258
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT300
set ibata_scale = 100
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set osd_vbat_pos = 2496
set osd_tim_2_pos = 2520
set osd_current_pos = 2503
set osd_mah_drawn_pos = 2511
set osd_avg_cell_voltage_pos = 2464
set osd_battery_usage_pos = 392

# profile
profile 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
Target: IFF4_AIO (iFlight SucceX-A F4 AIO)
Battery Meter Source: Onboard ADC
Battery: 5s/6s

fichek, 2020-05-20:
I'm having the same issue with two targets I tested with DJI (OMNIBUSF4SD and MATEKF411) and judging by the comments on Discord it seems like everyone has had that issue with DJI since the beginning.

I took a quick look at the code behind average cell voltage OSD element and there isn't much Emu does different from BetaFlight there, just slightly different filtering/rounding (which might be worth looking into since EmuFlight seems much more eager to round the number of cells up to 2 on HV 1s?).

So I dug a bit more and one diff in msp.c caught my eye: [EmuFlight](https://github.com/emuflight/EmuFlight/blob/c032327bc1e99baac006a59b34ce189895c59031/src/main/interface/msp.c#L583) vs. [BetaFlight](https://github.com/betaflight/betaflight/blob/2101326a1de4fa23e0a934bddeefefa2c18dedcb/src/main/msp/msp.c#L761) - could that be it? Why is BetaFlight adding 5 to the value of each voltage meter, then dividing by 10? I'm just checking this via browser as I'm traveling so can't track down the change, but I'd start there (and in the case above where BetaFlight writes `getLegacyBatteryVoltage()` in place of `getBatteryVoltage()` but then does `getBatteryVoltage()` too a few lines below). Disclaimer: I have no experience with flight firmware or MSP, just really want to see my average cell voltage again.

loutwice, 2020-05-20:
@fichek you completely right about it - i have been looking into code for a while and just received my hd goggle so i can try now -- there is also another difference in msp_analog : [Emuflight](https://github.com/emuflight/EmuFlight/blob/c032327bc1e99baac006a59b34ce189895c59031/src/main/interface/msp.c#L519)  vs [betaflight](https://github.com/betaflight/betaflight/blob/2101326a1de4fa23e0a934bddeefefa2c18dedcb/src/main/msp/msp.c#L698)

i tried to contact dji but no chance at the moment

fichek, 2020-05-20:
@loutwice great that you can take a look at this now! 🥳

Btw just realised my goggle dvr subtitles include correct values for both `uavBat` (total voltage) and `uavBatCells` (number of cells) so I don't think those are reported wrong over MSP and I don't see anything specific to average cell voltage in there, at least not named obviously. Maybe try asking BetaFlight devs? 🙈

loutwice, 2020-05-20:
@fichek  they dont like us - may be you can try :-D 
btw i dont thinf beta has done something for Dji but dji has adapt to it
i try to reach dji support but without success till now - still hoping for some answer

mbilozub, 2020-05-20:
HM. I compared BF, INAV, EMU and it seems I understood the problem. I have to try it...

mbilozub, 2020-05-20:
Yes, it worked for me. But I can't test on other targets or the analog osd.

That's what I changed:
```
Index: src/main/interface/msp.c
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- src/main/interface/msp.c	(revision 9366544a7d4a331953f29fcdf896c73739f20c22)
+++ src/main/interface/msp.c	(date 1589983163365)
@@ -525,6 +525,7 @@
         sbufWriteU16(dst, getRssi());
 #endif
         sbufWriteU16(dst, (int16_t)constrain(getAmperage(), -0x8000, 0x7FFF)); // send current in 0.01 A steps, range is -320A to 320A
+        sbufWriteU16(dst, getBatteryVoltage() * 10);
         break;
 
     case MSP_DEBUG:
@@ -563,6 +564,9 @@
 
         // battery alerts
         sbufWriteU8(dst, (uint8_t)getBatteryState());
+
+        // Additional battery voltage field (dji osd etc) (in 0.01V steps)
+        sbufWriteU16(dst, getBatteryVoltage() * 10);
         break;
     }
```

fichek, 2020-05-20:
@mbilozub great job! I can test on the two targets mentioned above in a couple of days when I get back home, but it  looks like a simple fix that should just work 👍

nerdCopter, 2020-05-20:
make it a Pull Request if so :)

mbilozub, 2020-05-20:
> make it a Pull Request if so :)

PR https://github.com/emuflight/EmuFlight/pull/242

loutwice, 2020-05-20:
@mbilozub thank you !! have some work in the gui now to make it release :-D

nerdCopter, 2020-06-09:
closing. merged PR #259

closed 2020-06-09 by nerdCopter