emuflight/EmuFlight #230: Baro didnt work on SucceX F7 TwinG BT Flight Controller

Whishp, 2020-05-18:
Baro didnt work on SucceX F7 TwinG BT Flight Controller https://shop.iflight-rc.com/index.php?route=product/product&product_id=930
EXF722DUAL/STM32F7X2 (S7X2)

nerdCopter, 2020-05-18:
@Kaiowarez

Kaiowarez, 2020-05-18:
I'll look into it, the testers versions had no baro so I prolly took it off
thinking it was not required..

Whishp, 2020-05-19:
Mby you need some logs?

Kaiowarez, 2020-05-19:
F7 twin g BT? Bluetooth? I never seen BT on iflight f7,
Dump helps from bf if you have em

But they have so many versions of the iff7 twin g fc one of then didn't
have a baro so I disabled it

Message me kaio Pryde in discord and I'll try to get it sorted out

Whishp, 2020-05-19:
I can't message you in discord. Add me Whishp#6055

nerdCopter, 2020-06-09:
non-BT version fixed and merged.
BT version, no tester report, so not merged.

closed 2020-06-09 by nerdCopter