emuflight/EmuFlight #245: Kakute F7 - No accel or gyro detected

GregMichalski, 2020-05-23:
Decided to give emuflight a try on my 3" cinewhoop (have had both Betaflight and Inav on it so far). Flashed the Kakute F7 and it takes an extraordinarily long time to boot up and be recognized by USB, and no accelerometer or gyro are detected.

`# status
System Uptime: 77 seconds
Current Time: 2020-05-23T06:02:52.088+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=216MHz, Vref=3.30V, Core temp=58degC, BARO=BMP280
SD card: Manufacturer 0x82, 15449088kB, 03/2014, v1.0, 'NCard'
Filesystem: Ready
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 4, config size: 2737, max available config: 32768
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Arming disable flags: NOGYRO RXLOSS CLI MSP

# version
# EmuFlight / KAKUTEF7 (KTF7) 0.3.1 May 19 2020 / 04:34:05 (9366544) MSP API: 1.47`

[image]
`

nerdCopter, 2020-05-23:
@Kaiowarez

Kaiowarez, 2020-05-23:
What board are you flashing exactly?
Kakute f7
Kakute f7 mini v1
Kakute f7 mini v2
Kakute f7 HD {dji version}

GregMichalski, 2020-05-23:
Kakute F7, the v1.5 specifically.
Not a mini. Not a HDV.

Kaiowarez, 2020-05-24:
I'll try to get you a fix soon, stupid kakutes keep changing every version, and emu still on hard coded targets..
Any chance you can do a dump from bf here, attach as a file if you can

GregMichalski, 2020-05-24:
Sure. Here you go.
[BTFL_cli_Taycan_KakuteF7_20200524_100051.txt](https://github.com/emuflight/EmuFlight/files/4673653/BTFL_cli_Taycan_KakuteF7_20200524_100051.txt)

udyux, 2020-05-24:
Came here to report the same issue. Flashed 0.3.2 to target kakutef7 (Holybro Kakute F7 V1.5). 

I'm guessing the FC isn't able to init properly which might explain the two additional issues I had. 

The ESCs don't chime to indicate successful FC connection, I only get the first three startup chimes. ESC is an FPVCrate 46A 32bit 4-in-1. It's a spare I had lying around while I wait for a replacement to ship (usually run a Holybro Tekko32 F3 40A 4-in-1).

The other issue was with the ImpulseRC ApexOSD board running the latest ImpulseOSD. I kind of expected it to have trouble as it hasn't always played nice with BF either but it works nonetheless. When I plugged in it simply couldn't detect the FC.

Kaiowarez, 2020-05-24:
You have the mpu6000 gyro right?
Try the kakutef7hd

udyux, 2020-05-24:
> You have the mpu6000 gyro right? Try the kakutef7hd
Correct and I will if there's a chance that the ApexOSD will recognize the FC. Unfortunately that's a bit of a dealbreaker for me for two reasons: the ImpulseOSD is awesome and I'd have to rewire everything in my quad to not use it.

The fact that it magically sets itself up with your FC, cam and VTx also means it's almost impossible to debug these things. I see it more as an issue on their end, not with EmuFlight.

Kaiowarez, 2020-05-24:
Hmm all we could hope for is that it works with it, but I know for a fact
the difference between kakutef7 and HD is the gyro is mpu6000 completely
same targets otherwise, and the regular kakutef7 is an Icm20689
Best thing I could advice if you have 1.5 is to use f7hd target and it'll
work I'm not too sure about the osd tho, and that auto detect feature on it
sounds nice.. I wonder if they'd design an fc with that built in on it

GregMichalski, 2020-05-25:
@Kaiowarez - I can confirm that the Kakute F7 HDV target does indeed work on the Kakute F7 v1.5 board. 

A few notes:

1. BF & INAV both recognize the MPU6000 on the v1.5 board and don't differentiate the Kakute F7 (ICM20689) from the Kakute F7 v1.5 (there is a Kakute F7 V2 target on BF but I don't see that board on the Holybro site anywhere, and INAV doesn't have an F7 V2 target) - I _**think**_ you addressed the behavior difference in your remark about hard coded targets but my coding skills and speak are ancient so if that's accurate ignore this whole entry :)
2. No access to VTX settings in configurator, however can access thru VTX Smart Audio on OSD so can be worked around for now until there is a unique target. OSD does work correctly (is Logo display on arming expected in future releases?)
3. I have an optical flow sensor (intended for use with INAV) connected to a UART which would normally communicate through the MSP protocol. Although I did not configure that UART for MSP EmuFlight flagged me on my first arming attempt for an active MSP connection. I was able to work around that by assigning that port to a RunCam peripheral to bypass this. Only mentioning it because it is different behavior than other FC software platforms (BF & INAV both both ignore it if not configured for MSP in the Ports tab.) I can enter that as a separate issue if desired/if it matters - as I'm still testing this rig and how I want to config it that isn't a "normal" issue I think.

I used the 3" 4S Cinewhoop preset (I'm actually @ 6S), briefly hovered and spun about in the living room on a 5S tester pack after config and did a couple flips and rolls. So far so good, will dig into my tuning over the next few days as time allows. Thanks - Greg

Kaiowarez, 2020-05-25:
Ah yes, unfortunately due to emuflight current state the devs was more
focused on getting flight code more stable than features
So the automatic gyro detection hasn't been updated to work like bf
standards yet.
So each target needs to be separated specifically still..
Eventually we will get there tho
And yes emu tends to autodetect what got plug/wired into what port/uart to
disable you can also just uncheck it or as you said slap is as a runcam
peripheral but I would definitely make sure it doesn't cause an issue for
your first before you send it out far and high if that was your plan

Kaiowarez, 2020-05-25:
Presets are awesome but it's meant as a starting point, just as a
precaution :)

If you need help tuning.. {not my expertise}
I suggest hoping on emuflight discord and asking in tuning channel
The emu community is very helpful in getting ppl sorted out the right way

TcheTcheFPV, 2020-05-26:
Hi,

I'm using a kakute F7 v1.5 and I confirm that I had to flash with the F7HDV target to get it working.

The only issue I have, and I don't know if it's related to the target, is the altitude showed in the OSD : it seems divided by 10. example: If the quadcopter is flying at 23m, it shows 2.3m.
I didn't find a command or option to fix that. Is there a setting available for that ?

Thank you.

nerdCopter, 2020-05-26:
@TcheTcheFPV  altitude fix has been made, but will be released in another release batch when we have more bugfixes.

TcheTcheFPV, 2020-05-26:
@nerdCopter thank you :)

nerdCopter, 2020-06-09:
Closing since "HD" works with "v1.5".  I think Kaio made some naming changes which will show with 0.3.2 release.
as for the features I think devs will work toward a 1.0.0 that would contain this.

closed 2020-06-09 by nerdCopter