emuflight/EmuFlight #246: Target specific GUI issues with presets

nerdCopter, 2020-05-23:
**Describe the bug**
W𝕙𝕀𝕥𝔼 🅗🅐🅣 -ҒᕈV-: mamba f411, mamba f405 and IFF411 have the issue
W𝕙𝕀𝕥𝔼 🅗🅐🅣 -ҒᕈV-: on my matek f722se that works
W𝕙𝕀𝕥𝔼 🅗🅐🅣 -ҒᕈV-: i think it's an issue about stm32f405 unified target
https://cdn.discordapp.com/attachments/637076600118050819/713793356639109130/20200523_183550_428x240.mp4

**To Reproduce**
select a preset, click save.  click another preset, click save.  

**Versions**
firmware 0.3.1; configurator 0.3.2

[EMUF_cli_20200523_203604.txt](https://github.com/emuflight/EmuFlight/files/4672346/EMUF_cli_20200523_203604.txt)
[EMUF_cli_20200523_202607.txt](https://github.com/emuflight/EmuFlight/files/4672347/EMUF_cli_20200523_202607.txt)

nerdCopter, 2020-06-02:
@Kaiowarez , any additional info this?

Kaiowarez, 2020-06-02:
I have not heard anything about this..

Kaiowarez, 2020-06-24:
New release fixed this @W𝕙𝕀𝕥𝔼 🅗🅐🅣 -ҒᕈV-?

nerdCopter, 2020-09-24:
same problem 2020-09-24 `[9:24 AM] Sinned_fpv: The fc is a mamba mini mk3`