emuflight/EmuFlight #264: quadcopter drift issue

TcheTcheFPV, 2020-05-28:
**Describe the bug**
While flying in accro mode, the quadcopter reacts as if I'm pitching back a little. It pitch back on it's own, until I need to pitch front to compensate. If I keep the pitch stick a little bit up, it stays stable. I have noticed the same behavior on the roll but with less impact.
I have checked the subtrim and Yaw/Pitch/Roll are at 1500 in the "receiver" configuration tab.
I have calibrated the accelerometer and problem is still there (i'm flying in accro mode, so even not calibrateed, it should not having any impact).

I decided to give a try to iNav, and the problem does not appear. I ask myself if this is not related to the fact that I have to flash my kakute F7 v1.5 with the kakuteF7HDV firmware (as flashing the kakute F7 v1.5 with the kakuteF7 firmware does not work) ?

**To Reproduce**
fly slowly in accro mode on a straight line and see that it's pitching back on it's own. May be related to the flight controller and the firmware.

**Expected behavior**
quadcopter should stay straight if not moving the sticks.

**Flight controller configuration**
# 

# diff all

# version
# EmuFlight / KAKUTEF7HDV (KTF7) 0.3.1 May 19 2020 / 04:35:01 (9366544) MSP API: 1.47

board_name 
manufacturer_id 
mcu_id 004f00283438511033343635
signature 

# reset configuration to default settings
defaults nosave

# name
name tche*tche

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beeper -RC_SMOOTHING_INIT_FAIL

# beacon

# map
map TAER1234

# serial
serial 2 2 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 5 1400 1600 0 0
aux 2 13 2 1400 1600 0 0
aux 3 26 3 1925 2100 0 0
aux 4 35 2 1925 2100 0 0
aux 5 36 1 1925 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 s 1000
rxfail 6 s 1000
rxfail 7 s 1000
rxfail 9 s 1000

# master
set gyro_lowpass_hz_roll = 60
set gyro_lowpass_hz_pitch = 60
set gyro_lowpass_hz_yaw = 60
set imuf_roll_q = 5000
set imuf_pitch_q = 5000
set imuf_yaw_q = 5000
set imuf_w = 128
set imuf_sharpness = 4000
set acc_calibration = 2,51,-31
set align_mag = CW270FLIP
set mag_calibration = 451,-225,128
set min_check = 1010
set max_check = 2000
set rssi_channel = 12
set serialrx_provider = CRSF
set airmode_start_throttle_percent = 15
set dshot_idle_value = 1500
set battery_meter = ESC
set ibata_scale = 182
set yaw_motors_reversed = ON
set 3d_deadband_throttle = 4
set gps_provider = UBLOX
set gps_auto_baud = ON
set deadband = 4
set yaw_deadband = 4
set osd_rssi_alarm = 60
set osd_cap_alarm = 1550
set osd_rssi_pos = 2105
set osd_tim_2_pos = 2529
set osd_vtx_channel_pos = 2082
set osd_crosshairs_pos = 2381
set osd_current_pos = 2535
set osd_mah_drawn_pos = 2543
set osd_craft_name_pos = 2090
set osd_gps_speed_pos = 2116
set osd_gps_sats_pos = 2100
set osd_home_dir_pos = 2444
set osd_home_dist_pos = 2446
set osd_compass_bar_pos = 2122
set osd_altitude_pos = 2132
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 2551
set osd_nheading_pos = 108
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = PAL
set vcd_v_offset = 5

# profile
profile 0

set dterm_lowpass_hz_roll = 80
set dterm_lowpass_hz_pitch = 80
set dterm_lowpass_hz_yaw = 80
set dterm_lowpass2_hz_roll = 0
set dterm_lowpass2_hz_pitch = 0
set dterm_lowpass2_hz_yaw = 0
set witchcraft_roll = 0
set witchcraft_pitch = 0
set pid_at_min_throttle = OFF
set spa_roll_p = 100
set spa_roll_i = 100
set spa_roll_d = 100
set spa_pitch_p = 100
set spa_pitch_i = 100
set spa_pitch_d = 100
set spa_yaw_p = 100
set spa_yaw_i = 100
set spa_yaw_d = 100
set throttle_boost = 0
set feathered_pids = 0
set emu_boost = 0
set emu_boost_yaw = 0
set emu_boost_limit = 0
set emu_boost_limit_yaw = 0
set p_pitch = 80
set i_pitch = 75
set d_pitch = 45
set p_roll = 70
set i_roll = 75
set d_roll = 40
set p_yaw = 75
set i_yaw = 85
set d_yaw = 8

# profile
profile 1

set throttle_boost = 0

# profile
profile 2

set throttle_boost = 0

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 195
set pitch_rc_rate = 195
set yaw_rc_rate = 195
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 44
set pitch_srate = 44
set yaw_srate = 44
set tpa_rate_p = 95
set tpa_rate_i = 155
set tpa_rate_d = 85
set tpa_breakpoint = 1400

# rateprofile
rateprofile 1

set roll_rc_rate = 180
set pitch_rc_rate = 180
set yaw_rc_rate = 180
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
Kakute F7 v1.5 (IMU MPU6000) 
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
crossfire RX / unify pro
7 inch quad with 2806.5 motors
 - how are the different components wired up.

**Additional context**

nerdCopter, 2020-05-28:
I think some drifts are being found on other FC too. it's not 100% validated/clear at this moment, but reports for yes-drift and no-drift have both been received.  now the idea is to find source of issue.  Thank you for the submission!

TcheTcheFPV, 2020-05-28:
@nerdCopter I can record a blackbox log if you want.

nerdCopter, 2020-05-28:
That could help.
Also consider setting quad on solid floor, stabilize the usb cord, and watch sensors tab in configurator (expert mode). Enable gyro and accelerometer.  let it read sensors for 5 minute or more and screenshot the graphs at that point.
It does not drift in BF?
Thank you.

TcheTcheFPV, 2020-05-29:
Here is log files and screenshot.
[debug_264.zip](https://github.com/emuflight/EmuFlight/files/4702804/debug_264.zip)

I didn't try BF, only iNav, and no drift with iNav.

nerdCopter, 2020-05-29:
* i was only able to open log 3 & 6
* it's very hard to see drift, but i looked at RC Command for straight-flight.  a pro (not me) should be able to see any drift. (maybe set rc command zoom to 105% of graph setup, you know you straight-flights better as a pilot, then me as an observer.)
* I am unsure what version 1.5 is.  we have v1 and v2 Kakute's, where "HD" implies "DJI", and the difference in code on these targets are:

v1: 
#define ACC_1_ALIGN              ACC_ICM20689_ALIGN
#define GYRO_1_ALIGN             GYRO_ICM20689_ALIGN
#define USE_SPI_DEVICE_4   //ICM20689

v2: 
#define ACC_1_ALIGN              ACC_MPU6000_ALIGN
#define GYRO_1_ALIGN             GYRO_MPU6000_ALIGN
#define USE_SPI_DEVICE_4   //MPU6000

If you love/happy with your tune, ignore this, but further potential tuning:
Possibly reduce dTerm (Kd) by one or two points on roll,
 and two or three on pitch.
```
set gyro_lowpass_hz_roll = 90
set gyro_lowpass_hz_pitch = 90
set gyro_lowpass_hz_yaw = 80
set dterm_lowpass_hz_roll = 95
set dterm_lowpass_hz_pitch = 95
set dterm_lowpass_hz_yaw = 85

# optional stuff if not racing
set dterm_lowpass2_hz_roll = 180
set dterm_lowpass2_hz_pitch = 180
set dterm_lowpass2_hz_yaw = 180
# rc smoothing:
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set rc_smoothing_input_hz = 50
set rc_smoothing_derivative_hz = 75
set rc_smoothing_input_type = BIQUAD
set rc_smoothing_derivative_type = BIQUAD
```

nerdCopter, 2020-05-29:
@Kaiowarez , can you clarify which target he needs for kakute f7 v1.5?

Kaiowarez, 2020-05-29:
Kakutef7hdv for v1.5 because of the mpu6000 sensor

TcheTcheFPV, 2020-05-30:
Hello and thank you for your analyse and tuning tips @nerdCopter, I will give a try next time for sure. I'm wondering if it can be related to the filtering settings (too much noise to the gyro can make the quad drifting ?)

I realize now that the log itself might be unclear, so please find the recorded video flight that you can open on the log to see what I mean with drifting : https://we.tl/t-xGwh5qnGry

in the video you will see :
-> at 1:23 : flying straight forward, and the quad start drifting on roll left.
-> at 1:49 / 2:05 / 3:41 : stabilize the quad and it start drifting pitch back

nerdCopter, 2020-05-30:
* the confusing part is some people are having this and other are not.
* some people resolve with iTerm (Ki) and others not.
* maybe some wrong math has been programmed that we cannot source, or
* maybe a driver for some specific hardware is incorrect.
* We do know about it, but 
* another task-in-the-works may actually override this problem.
* i am unclear if this will get further attention from dev(s) or if the other task takes precedence.
* this issue-submission will remain open.

nerdCopter, 2020-06-02:
@TcheTcheFPV , could you set the following and report if this changes your drift behavior?
```
set spa_roll_i = 100
set spa_pitch_i = 100
set spa_yaw_i = 100
```

TcheTcheFPV, 2020-06-02:
Hi @nerdCopter. I will not be able to test before a week or two. But as soon as I can test that, I will let you know.

Thank you.

nerdCopter, 2020-06-19:
i've had some more news:
* you might try reducing iDecay.
* if you do not need accelorometer for angle/nfe/horizon, consider testing with accelorometer off.

TcheTcheFPV, 2020-06-25:
Hello @nerdCopter ,

Sorry for my late answer, I could not test anything before today. So, I tried all the advice you put here, one by one.

So, first of all, thanks for the tuning advice on the filtering: it gives a better behavior to the quadcopter. I wish I could find some documentation/how to on how to analyse the blackbox output and adjust filtering, pids and so on, as you did here. Is there some existing ?

But, this didn't solve the issue, as it was still here, the quad was drifting the same, no more, no less. Disabling the accelerometer also didn't help.

But, when I have reduced the iDecay value to the minimum (1), the drift issue was gone ! At the beginning, it was 4 and I have reduced it to 2, and the quadcopter was still drifting a little. So, I decided to go to the minimum, and that was it !

At the end, I don't know why the iDecay value act like that. On my other quad (5 inch), I have an iDecay value of 6, and it does not drift at all.

The quadcopter is now flying very well, I'm very happy with EmuFlight and that tune :)

Thanks again for your help !

nerdCopter, 2020-06-25:
@TcheTcheFPV this is good feedback, thank you very much & confirming some recent finding from myself (although i stopped iDecay at `2` and should have tried `1`). 
We have known some peculiarities with iTerm lately, and this may lead to some additional finding, so it is very appreciated.

gretel, 2020-07-12:
why do there have to be three ways to mess with `I`?
it should all be disabled by default. the value of `0` should actually disable it, not `1`.
at the same time, new features are being added, while this has never been working right.

nerdCopter, 2020-07-16:
Drift will be fixed with https://github.com/emuflight/EmuFlight/pull/320 @TcheTcheFPV

TcheTcheFPV, 2020-07-16:
Nice, thank you :)

closed 2020-07-29 by nerdCopter