emuflight/EmuFlight #281: SFHSS spi protocol issue

djakette123, 2020-06-09:
**Describe the bug :**
When i try to change the protocol to SFHSS in the cli, board reboot and is no more recognized by GUI. It's with a mobula6 crazybeef4fr.

**To reproduce**
You can flash 0.3.1 to CRAZYBEEF4FR, change the spi protocol to SFHSS, save and board will not connect anymore to gui.

nerdCopter, 2020-12-14:
is this fixed with configurator_0.3.4?, we added it awhile back and would like to close the ticket if it works proper.

nerdCopter, 2020-12-15:
`FR` implies FrSky.  is there something else i'm misunderstanding here?

djakette123, 2020-12-15:
Frsky spi supports SFHSS since betaflight 4.1
I know my stuff.
I'm currently flying bf 4.2 on it with my futaba...

Quick-Flash, 2021-04-02:
are you able to use SFHSS if you do it in the cli?

djakette123, 2021-04-02:
> are you able to use SFHSS if you do it in the cli?

Yeah i did try with cli it's the same, when i save the board don't show COM port again.
I tried other thing, switching from FRSKY_X to FRSKY_D works fine but that's all.
When i set SFHSS it failed, it's the same if i set to flysky protocol board would not connect again.

djakette123, 2021-06-10:
any update after 1 year ?

nerdCopter, 2021-06-10:
no. i think no effort on this will take place.
After 0.4.0 release, all efforts are re-routing to Emuflight 1.0.0 which is rebased on betaflight 4.x and will inherit working sfhss.

djakette123, 2021-06-10:
ok thanks for the update