emuflight/EmuFlight #288: No Emuflight firmware for the Transtec Beetle transtecf411hd Board

f3fphil, 2020-06-19:
Would Love to fly Emuflight on my Beetle but there is not the Right Firmware the STM32F411 Says Nogyro ....

Kaiowarez, 2020-06-19:
> Assigned #288 <https://github.com/emuflight/EmuFlight/issues/288> to
> @Kaiowarez <https://github.com/Kaiowarez>.

Noo don't assign to me @nerdCopter lol

Anyway
Post a link to the fc and a dump thanks

f3fphil, 2020-06-20:
https://www.transtechobby.com/products/transtec-f411-hd-for-dji-fpv-system

f3fphil, 2020-06-20:
What do i habe to copy for the Dumb ?
Thx for your help

Kaiowarez, 2020-06-20:
Connect the fc to bf configurator
Go to cli last tab type dump
Then on lower right of the window you'd see a button save to file save to
file name it whatever you want and attach file here or post on target
request in discord works as well

Thanks

f3fphil, 2020-06-20:
Ok thx a lot i Send it at Monday

Kaiowarez, 2020-06-20:
No problem

f3fphil, 2020-06-21:
[BTFL_cli_TransTEC_20200621_203827.txt](https://github.com/emuflight/EmuFlight/files/4810057/BTFL_cli_TransTEC_20200621_203827.txt)

f3fphil, 2020-06-21:
theres the Dumb

Kaiowarez, 2020-06-24:
hello can you join emu discord for testing?  https://discord.gg/vVtVKP3

f3fphil, 2020-06-24:
I m on discord 
Username F3fphil

f3fphil, 2020-06-24:
Im sorry but i dont no what i have to Search for on Discord 🙈

nerdCopter, 2020-08-10:
@f3fphil , did you ever receive the `transtecf411hd` `.hex` ?

MorenoMdz, 2020-09-17:
Hey, do we need testing for this still? I just got this model and wanted to give emuflight a try for indoor flying

Kaiowarez, 2020-09-18:
Hop on discord maybe someone got it available but it's fully tested

nerdCopter, 2020-09-29:
https://github.com/emuflight/EmuFlight/issues/288
https://github.com/emuflight/EmuFlight/pull/318
https://github.com/emuflight/EmuFlight/pull/365

closed 2020-09-29 by nerdCopter