emuflight/EmuFlight #300: RC smoothing Filter and Auto not working

ghden, 2020-06-28:
Using Filter for RC Smoothing, tested with frsky R-XSR and R9 mini on two separate quads, only thing in common was Helio Sprint V2 and StrixF10 on another.. I didn't tried non IMUF boards.

Same result with 0.3.x and older versions.

RC Smoothing set as Filter and Auto, and all defaults, the RC signal is not smooth it's very digital square lines.. making weird motor sound during rolls and flips, hot motors, etc.  

Workaround: Set to manual and put any value grater than 0, I didn't try to put 0 actually because in theory is the same as auto..  Actually best result is around 50-60, didn't tried higher values.. but anyways but apparently auto is working as feature disabled.

Setup 1:
Helio Spring V2

Setup 2
StrixF10

This issue very easy to reproduce and not related to VTX, Receiver, ESC, Motors.. That's why not adding the settings.. Anyways if required I can send black box examples. and anything you guys need.

Thanks!

ghden, 2020-06-28:
I'm so sorry, forgot to mention something important, I using FPORT, not tested any other protocol like SBUS.. I have no receiver with SBUS firmware and not wanted to change the firmwares.

Vanalio, 2020-06-28:
I can confirm, auto is not working well. With FPORT I had somehow acceptable results with interpolation at 7ms.

nerdCopter, 2020-06-29:
* this was never changed, this was inherited from butterflight.
* `auto` never worked well. on paper `auto` seems good, but in practice always reported to be bad.
* recommendation is similar to https://github.com/emuflight/EmuFlight/wiki/RC-Smoothing ; seems 50/75 is good.

ghden, 2020-06-29:
Sounds good to me, in that case Auto should not be the default when we set Filter, and then instead of 0 can be the default values 50/75 or so we know works well and problema solved! There are many pilots never look at black box, if they are no novice, then can feel that is not flying well and may think is because Emuflight is not good.. like I was thinking initially but I never give up and used BB to troubleshoot. Betaflight Auto works very well... you guys can put back the Betaflight code was changed on Butterflight.. (Maybe??)

nerdCopter, 2020-06-29:
* i dont think butterflight actually changed it. i think butterflight inherited from betaflight 3.5.x, but i could be wrong.
* we have discussed defaults, but never transpired.  i will mention it again.

nerdCopter, 2020-06-29:
PR #302 made.

closed 2020-10-18 by nerdCopter