emuflight/EmuFlight #301: Matek 5.8G VTX-MINI cli config problem

DarkSoulRL, 2020-06-29:
EmuFlight / MAMBAF722 (MBF7) 0.3.1 Jun 27 2020 / 22:21:57 (0c240c226) MSP API: 1.48

This VTX has 5 power levels, but only 4 can be set in the CLI. It is configured normally through the OSD.

[image]

[image]

nerdCopter, 2020-06-29:
0 is a level, should be 25mw
0,1,2,3,4 = 5 levels.
now if it's really 6 levels, with 5 being the last, then maybe @Quick-Flash can fix it.
or if it's not, then the code need to be fixed to not allow `5`.

DarkSoulRL, 2020-06-29:
Default Value 1 
In fact, 1 = 25 mW, which means 0, I do not know, I have never used it.

nerdCopter, 2020-06-29:
@DarkSoulRL , Thank you.

@Kaiowarez , care to revisit this issue? :P -- is this target specific?
 or do we need to modify common code @Quick-Flash ?

i see the following definitions in  `src/main/drivers/vtx_common.h`: smartaudio goes to 4 and tramp goes to 5. unsure why he cannot get 5 -- i don't have Tramp to test.
```
// SmartAudio "---", 25, 200, 500, 800 mW
#define VTX_SA_POWER_OFF          0
#define VTX_SA_POWER_25           1
#define VTX_SA_POWER_200          2
#define VTX_SA_POWER_500          3
#define VTX_SA_POWER_800          4

// Tramp "---", 25, 100, 200, 400, 600 mW
#define VTX_TRAMP_POWER_OFF       0
#define VTX_TRAMP_POWER_25        1
#define VTX_TRAMP_POWER_100       2
#define VTX_TRAMP_POWER_200       3
#define VTX_TRAMP_POWER_400       4
#define VTX_TRAMP_POWER_600       5
```
are we just out-dated and there is an easy fix? or is this the whole vtx tables can-of-worms?

i see: 
`src/main/interface/settings.c:1050:    { "vtx_power",                  VAR_UINT8  | MASTER_VALUE, .config.minmax = { VTX_SETTINGS_MIN_POWER, VTX_SETTINGS_POWER_COUNT-1 }, PG_VTX_SETTINGS_CONFIG, offsetof(vtxSettingsConfig_t, power) },` so it obviously calcs the max.

Kaiowarez, 2020-06-30:
I checked the code as well looks right logically
It might be a port setting issue not change the vtx config probably.. 
And I been soo busy that I get lucky to have time to finish half a target :(

DarkSoulRL, 2020-06-30:
[image]
cant set 800mw  in CLI

gretel, 2020-07-10:
`tbs` defines it's standard using 5 levels of power, while `immersionrc` defines 6 levels of power. it's all fine.
but, the vendors do not stick to these proprietary standards and do whatever they feel is fine.

the actual power levels of this particular vtx are

```
PIT, 25, 200, 400, 800mW
```

as described at http://www.mateksys.com/?portfolio=vtx-mini#tab-id-2

closed 2021-05-25 by nerdCopter

nerdCopter, 2021-10-05:
#705