emuflight/EmuFlight #316: Kakute F7: motors won't spin

Emanuele-Spatola, 2020-07-10:
**Describe the bug**
Motors won't spin from the motor tabs, when plugging the battery the motors don't play the second tone after the ESC initialisation.
I then reinstalled BF 4.2 and all the motors are spinning correctly.
Also with Emuflight installed the BlHeli32 Suite cannot read the ESC configuration.

**To Reproduce**
Plug the battery, just the first 3 tone will play then nothing.

**Expected behavior**
A second beep is played.

**Flight controller configuration**
Stock configuration.
# version
# EmuFlight / KAKUTEF7 (KTF7) 0.3.1 May 19 2020 / 04:34:05 (9366544) MSP API: 1.47

**Setup / Versions**
 - Kakute F7 v1.5
 - Flycolor x-cross 4 in 1 ESC

nerdCopter, 2020-07-10:
For v1.5 you can use the `KAKUTEF7HDV` but i think there was fixes since the release.  Are you in Discord?

gretel, 2020-07-12:
`f7 mini v1` fine for me.

Emanuele-Spatola, 2020-07-14:
Used the `KAKUTEF7HDV` target provided by @nerdCopter and it worked.
Maybe it could be an good idea to have a "Kakute 1.5" option in the configurator that uses the `KAKUTEF7HDV`?
I could try to do it if you too busy (no promises on the result XD)

nerdCopter, 2020-07-16:
@Emanuele-Spatola what ID does the 1.5 show from betaflight? i.e. `KTF7` or something else?

closed 2021-05-25 by nerdCopter