emuflight/EmuFlight #317: gps speed

loopingfpv, 2020-07-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

loopingfpv, 2020-07-11:
salut, 
je suis devant un fait assez désagréable, en installant mon module gps avec emuflight et en activant la le gps speed, la vitesse est en cm/sec serait il possible de modifier pour indiquer la vitesse en km/h. Aussi, dans l'onglet osd il manque la case flip arrow comme dans betaflight
merci

nerdCopter, 2020-07-11:
google translation: 
```
Hi,
I am faced with a rather unpleasant fact, by installing my gps module with emuflight and by activating
 the gps speed,  the speed is in cm / sec it would be possible to modify to indicate the speed in km / h. 
Also, in the tab osd it misses  the box flip arrow as in betaflight
thank you
```

@loopingfpv , I do not personally know about the `km/h`.
But i think you are needing the expert-mode in OSD:
[image]

loopingfpv, 2020-07-11:
Hi... I'm in expert mode. But on my osd it is in cm/secEnvoyé depuis mon téléphone Huawei-------- Message original --------De : nerdCopter <notifications@github.com>Date : sam. 11 juil. 2020 à 18:32À : emuflight/EmuFlight <EmuFlight@noreply.github.com>Cc : loopingfpv <dbutera@skynet.be>, Mention <mention@noreply.github.com>Objet : Re: [emuflight/EmuFlight] gps speed (#317)
google translation:
Hi,
I am faced with a rather unpleasant fact, by installing my gps module with emuflight and by activating
 the gps speed,  the speed is in cm / sec it would be possible to modify to indicate the speed in km / h. 
Also, in the tab osd it misses  the box flip arrow as in betaflight
thank you

@loopingfpv , I do not personally know about the km/h.
But i think you are needing the expert-mode in OSD:

gretel, 2020-07-12:
https://github.com/emuflight/EmuFlight/blob/fb696a2482e5bb4eb4f173e8011e7d8d427f9ea2/src/main/io/osd.c#L503

closed 2021-05-25 by nerdCopter