emuflight/EmuFlight #319: BUG on  GPS

stewefpv, 2020-07-13:
I found two problems on the use of the gps, ( matek gps on 2 quads, and a TBS gps) the altitude does not work, and the up / down speed variation does not work

stefano

AndroidGX, 2020-07-16:
Hello,
I confirm that GPS Altitude is not working for me too (FC: "MAMBAF405US")
The arrow pointing from home direction is not working too.
Cheers'

nerdCopter, 2020-07-16:
i think this was fixed in May and ready for next release (hopefully soon): https://github.com/emuflight/EmuFlight/pull/259

dog4fly, 2020-07-31:
I have also two problems with the usage of GPS  ("BETAFLIGHTF4"  FC and BN-180 GPS)
1. the BN-180 works only with the ublox protocol and not with NMEA as usualy and as with Betaflight 4.1 before.
2. the displayed value for the GPS altitude is abt. factor 10 (estimated) to low.  For examble:  in a estimated altitude of abt. 85m a value of 8.5m is displayed-

I don't know, whether this issues are known already or not.  Or do I make a mistake in the configuration of Emuflight 0.3.1 ?   Thanks in advance for your help!
Peter

dronotron, 2020-08-09:
I can confirm the issue on the shown altitude. It shows right on the initial fix. But after arming (when it gets "zeroed") it shows very small values no mater how high the quad gets. I'm using a Helio Spring IMUF V1

nerdCopter, 2020-08-10:
I recall "something" was fixed for altitude, but not yet released.  however, i don't think much effort was put into GPS yet.  Emu is a fork of Butterflight which was a fork of Betaflight 3.5.x.
EmuFlight's focus has been on acro flight control and noise filtering features thus far.
If you guys want to join Discord, we can offer you a latest compile .hex of master source-code.
https://discord.gg/gdP9CwE

nerdCopter, 2020-10-18:
is altitude working properly on 0.3.2 for you guys?

AndroidGX, 2020-10-18:
> is altitude working properly on 0.3.2 for you guys?

yup! everything is working well (at least, for me)

dronotron, 2020-10-19:
> is altitude working properly on 0.3.2 for you guys?

Actually, it worked on the build with the dynamic filter working you sent me a while ago. I didn't realized the altitude was fixed there until yesterday 😅 I haven't updated to 0.3.2 yet

closed 2020-10-19 by nerdCopter